arendst/Tasmota #2456: Sonoff 4CH Pro R2 toggle switch issues

TheCellMC, 2018-04-16:
I'm having a few issues with my Sonoff 4CH Pro R2.
First of all, the toggle switches don't seem to change functionality depending on the SwitchMode. I have tried setting them in the `userconfig.h` and via the `MQTT command`. The SwitchMode changes but not the functionality of the toggle switch.
Secondly, if I press many times on one of the toggle switches, they get disabled for a few minutes.
Can these problems be solved?

Leseratte10, 2018-04-16:
The second thing is due to the RF thing - pressing a button twice in a row activates the RF (433 MHz) learning mode to pair a 433 MHz remote to the corresponding channel. This can not be disabled as this is not handled by the ESP, but by another dedicated RF chip. 

What behaviour are you trying to achieve with the Switchmode? If I recall correctly, the switchmode is intended to be used with external switches (such as on GPIO14), not on the internal ones, but I might be wrong there.

nmaggioni, 2018-04-16:
@Leseratte10 Do you have any suggestions on how to disable RF features via hardware (cutting traces, etc.)?

I don't use them and if I happen to mis-doubleclick the button connected to the relay it does not only disable the relay for a few minutes, it also often learns some random RF noise and randomly toggles my lights on and off.

TheCellMC, 2018-04-16:
If I swap my Sonoff 4CH Pro R2 with a Sonoff 4CH R2, will that solve the RF thing as that board does not have RF functionality if I am correct.

So I can't change the Sonoff 4CH Pro R2 of the internal pins? Currently the pins have the following behaviour - it's a sort of mix of momentary and two way buttons. If I press the button quickly, the relay toggles state. However, if I hold the button, it behaves as a two way switch where, by closing the button, the relay is closed and by opening the button, the relay is open.
I'm connecting these to wall switches so all I need is two-way behaviour. The problem with the current behaviour is that, if I toggle the switch quickly, the light stays on as it behaves as a momentary button when I quickly open and close the button. Secondly, this behaviour seems to come with a delay (0,5s). I'm guessing that the delay is there for the board to see if it needs to behave as a momentary or two-way switch.

closed 2018-04-17 by TheCellMC