arendst/Tasmota #2458: Sonoff Dual R2 MQTT only one cmnd POWER

BigWumpus, 2018-04-16:
Hallo,
habe heute meinen Sonoff DUAL R2 bekommen und TASMOTA 5.12.00 mit ATOM.IO geflasht.
Es gibt in den Einstellungen zwei Felder für "friendly name" mit "Sonoff" und "Sonoff2" vorbesetzt. Wozu?
Schlimmer ist aber, daß im ioBroker im MQTT-Adapter nur ein cmnd-Datenpunkt "POWER" auftaucht, der nur das erste Relais schaltet. im stat-Zweig gibt es erwartungsgemäß POWER1 und POWER2.
Ich habe/mußte die Ansteuerung jetzt auf http-Requests umbauen.

Leseratte10, 2018-04-16:
Friendly name ist dafür da, wenn du den Sonoff mit Alexa ansteuerst, um den beiden angeschlossenen Geräten einen Namen geben zu können. 

Tauchen die POWER-Einträge im iobroker auf, wenn du einfach mal einen MQTT-Befehl dahin schickst?

BigWumpus, 2018-04-17:
Nein, die Einträge erscheinen auch nach der Benutzung (über Knöpfe oder URL-Requests) nicht auf.

Leseratte10, 2018-04-17:
Nicht über Knöpfe oder URL. Über MQTT. Unter Linux z. b. mit `mosquitto_pub -t "cmnd/sonoff/POWER2" -u DVES_USER -P DVES_PASS -m true -h 127.0.0.1 -i client -p 1883`

(Username, Passwort und IP evtl. anpassen)
Obwohl das prinzipiell auch ohne funktionieren müsste ...

BigWumpus, 2018-04-17:
Ich habe im ioBroker die beiden topics für cmnd/.../POWER1 und 2 per Hand angelegt, nach dem existierenden Eintrag ...POWER. Ich kann sie mit ON und OFF beschreiben und die bereits vorhandenen stat-Topics gegen entsprechende Rückmeldung. Der Schalter selbst liegt zu Hause auf dem Schreibtisch.

BigWumpus, 2018-04-25:
Ach ja, die stat/.../POWER1 und ...POWER2 sind jetzt ohne Updates und werden wohl nicht mehr gemeldet.

closed 2018-06-24 by stale[bot]