arendst/Tasmota #2461: 5.12.0l Sunrise and Sunset Times are Way Off

dennyfmn, 2018-04-16:
Hardware (2) Sonoff TH with Si7021 sensor, also Sonoff Basic with added DS18B20 sensor. All three units report the same erroneous sunrise and sunset times.

Result of Status 0, notice sunrise and sunset times:

stat/sonoff06/STATUS {"Status":{"Module":4,"FriendlyName":["Sonoff06"],"Topic":"sonoff06","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
stat/sonoff06/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.3:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"1T01:05:50","StartupUTC":"2018-04-15T21:15:17","Sleep":0,"BootCount":12,"SaveCount":38,"SaveAddress":"FA000"}}
stat/sonoff06/STATUS2 {"StatusFWR":{"Version":"5.12.0l","BuildDateTime":"2018-04-15T16:08:37","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
stat/sonoff06/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":1,"LogHost":"192.168.1.29","LogPort":514,"SSId":["microwap2",""],"TelePeriod":60,"SetOption":["00008129","55818000"]}}
stat/sonoff06/STATUS4 {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
stat/sonoff06/STATUS5 {"StatusNET":{"Hostname":"sonoff06-2920","IPAddress":"192.168.1.104","Gateway":"192.168.1.240","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.240","Mac":"2C:3A:E8:4F:4B:68","Webserver":2,"WifiConfig":3}}
stat/sonoff06/STATUS6 {"StatusMQT":{"MqttHost":"piserver","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4F4B68","MqttUser":"DVES_USER","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff06/STATUS7 {"StatusTIM":{"UTC":"Mon Apr 16 22:21:07 2018","Local":"Mon Apr 16 17:21:07 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"17:38","Sunset":"31:10"}}
stat/sonoff06/STATUS10 {"StatusSNS":{"Time":"2018-04-16T17:21:07","SI7021":{"Temperature":75.4,"Humidity":41.8},"TempUnit":"F"}}
stat/sonoff06/STATUS11 {"StatusSTS":{"Time":"2018-04-16T17:21:07","Uptime":"1T01:05:50","Vcc":3.504,"POWER":"OFF","Wifi":{"AP":1,"SSId":"microwap2","RSSI":100,"APMac":"10:C3:7B:D1:FE:D0"}}}

Latitude and Longitude are set correctly:

stat/sonoff06/RESULT {"Latitude":"44.736330"}
stat/sonoff06/RESULT {"Longitude":"-93.308710"}

Thanks!

localhost61, 2018-04-17:
Yes indeed, your daytime is more than 24h in the USA, what a chance, you're able to do more work in a day ;-).

Try with the absolute value of the Longitude, it may solve the issue.

Even if this is not the source code origin, I've seen a warning [in this git](https://github.com/buelowp/sunset) that a negative Longitude leads to odd results.
(In Tasmota's xdrv_09_timers.ino @arendst  writes that he refers to [Sonnen Auf- und Untergang mit dem Arduino berechnen](https://forum.arduino.cc/index.php?topic=218280.0) )

localhost61, 2018-04-17:
Yep, the code is not tuned.

Playing a bit with longitude and latitude commands in the Webserver console, it appears that negative numbers are not properly stored for both Longitude and Latitude, clearly a data typing error.
`13:20:43 CMD: longitude -2.48
13:20:43 RSL: RESULT = {"Longitude":"-1.520000"}
13:24:02 CMD: latitude -8.00001
13:24:02 RSL: RESULT = {"Latitude":"-7.999990"}
13:24:24 CMD: latitude -1.48
13:24:24 RSL: RESULT = {"Latitude":"-0.520000"}`
The mantisse sign is opposite of the integer:
`-1.48 -> (-1)+(0.48) = -0.52 
-93.308710 -> (-93)+(308710) = -93.691290`
It's not the same error but it shows where to search ;-)

dennyfmn, 2018-04-17:
Thanks for your responses!

 

I read the src page you linked and found it interesting. I don’t understand how removing the minus sign would help, since it would point to a location east of the prime meridian instead of west.

 

I did remove the minus sign from my latitude just to try it, and recompiled. I got different times for sunrise and sunset that were much better but still do not agree with actuals.

 

mosquitto_pub -h piserver -t cmnd/sonoff06/STATUS -m 7

{"StatusTIM":{"UTC":"Tue Apr 17 12:58:26 2018","Local":"Tue Apr 17 07:58:26 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"05:10","Sunset":"18:45"}}

 

mosquitto_pub -h piserver -t cmnd/sonoff06/Longitude -n

{"Longitude":"93.308710"}

 

Weather Underground reports:

                                                                Rise                        Set

Actual Time                              6:26 AM           8:01 PM

Civil Twilight                          5:55 AM           8:31 PM

Nautical Twilight                    5:18 AM           9:09 PM

Astronomical Twilight           4:38 AM           9:48 PM

 

NOAA Reports for Minneapolis DST:

Latitude                                               44.97

Loggitude                                            -93.25

Apparent Sunrise                             06:25

Apparent Sunset                              20:01

 

So, our calculated values, Sunrise 05:10 and Sunset 18:45, vary quite a bit from the actuals.

 

It would appear that calculating sunrise and sunset from position and date removes the necessity of having an Internet connection but is quite complicated and not very generalized for all longitudes.

 

If you can assume an Internet connection, maybe it would be easier and more accurate to just look it up.

dennyfmn, 2018-04-17:
> @dennyfmn <https://github.com/dennyfmn>  there might be more things wrong but for now the acceptance of your longitude/latitude inputs fail because I used a routine to keep code space small but it seems not to be minus friendly.
>
> I'll have to fix that first before going on to a then possible location problem.
>
> Stay tuned.

Great! Thanks for you response and cool firmware!

arendst, 2018-04-17:
Fixed the negative value issue. Pls enter your values again and see if it finds the correct Sunrise/Sunset values.

dennyfmn, 2018-04-17:
Thanks! I’ll give it a try when I get back home and let you know what happens.

 

How many settings can I put into the user_config override? The sample just shows a few of the main settings.

 

I’d like to include all the defines I have commented out to save memory in the override file if possible. That will save me a lot of time when I update Tasmota to a new version. I have not taken advantage of this feature yet.

arendst, 2018-04-17:
Fixed calculation sunrise and sunset for Minneapolis and the world ;-)

You can add as many defines as you like in the override file. Just give it a try.

dennyfmn, 2018-04-17:
Great! I’ll check it out when I get home…

dennyfmn, 2018-04-17:
I’m happy to report that the sunrise and sunset time are now good for negative longitudes. For my location near Minneapolis MN USA on 4/17/2018:

 

Sunrise 06:24

Sunset  19:59

 

These times are very close to the ones posted by Weather Underground and NOAA.

 

Thanks for your quick response and fix!

 

I am a bit confused about user_config_override.h. I tired using it but no matter how many times I changed CFG_HOLDER in both user_config.h or user_config_override.h, when compiling, uploading to the OTA server, and doing a firmware update from the OTA server, the section 1 parameters were not changed. I finally commented back out

 

//#define USE_CONFIG_OVERRIDE           in user_config.h

 

Then manually made all my local changes to the new downloaded user_config.h, compiled and updated three Sonoffs. I had to update the configurations manually through the web interface as expected and the sunrise/sunset fix was finally confirmed.

 

I realize this is getting a little off topic but can anyone explain why user_config_override.h did not work as expected?

 

Thanks!

localhost61, 2018-04-17:
Perfectly working here too after firmware update. Same values as on [sunrise-and-sunset](http://www.sunrise-and-sunset.com/fr/sun/france/rennes).
2018-04-18
Sunrise: 07h10
Sunset: 21h00

[image]

It seems mandatory to specify days 1111111 otherwise it seems to do nothing. I'll see in a few hours.

closed 2018-04-18 by arendst

gorgiea, 2018-04-18:
Is it possible to display the current time and timezone on this page?.. makes it easier to detect issues. Maybe even have a web interface to set the time/timezone?

gorgiea, 2018-04-18:
Also the offset should be labeled.. some might get confused on what these boxes are for or which one is hh:mm.

[image]

arendst, 2018-04-18:
See 5.12.0m where it makes more semse I think.

gorgiea, 2018-04-18:
Ok 5.12.0m is better... but now it seems I set the sunset timer on the "Timer Config" page and when I go to check it in the console, I run the "timer 1" cmd which just shows me that the sunset time is "00:00", I run the "timer 1" cmd again, and it resets my timer to "Mode 0"... Is this expected? Should it reset the timer just from running the "timer 1" cmd? Also should the sunset time show something other than: "00:00", shouldn't it show "20:04" (my sunset time) in the console?

[image]

[image]

localhost61, 2018-04-18:
The command is "timer1" to retrieve Timer #1 data. Remove the blank.

gorgiea, 2018-04-18:
Ok, so just by having the space there, it was copying timer3 to timer1?...

Will the sunset and sunrise "Time" always show as "00:00" in the console? 

[image]

localhost61, 2018-04-18:
Check Latitude, Longitude and insure Timezone = 99

localhost61, 2018-04-18:
Check Latitude, Longitude and insure Timezone = 99

And yes, if Timer is an alias for Timer1, with the command : Timer 3 you made a copy, if I refer to the [Commands page](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

gorgiea, 2018-04-18:
Checked latitude and longitude, also set timezone to 99. Still shows "00:00" as the "Time" for sunset "timer1" .

We'll see if the sunset timer1 works here, shortly... (EDT, sunset = 20:04). 

Thanks for your help!

[image]

localhost61, 2018-04-19:
Same behavior here (regression???).
I'll know more about it a couple of hours ;-)

arendst, 2018-04-19:
It's not showing sunset or sunrise time. It's the offset time what is shown equal to the webpage.

localhost61, 2018-04-19:
@arendst  I think that something is broken now in the timers behavior with v5.12.0m.
I'm playing with rules, timers ant sunrise. And when testing standard "Time" timers it appears that if the timer is properly stored, the hour field is reset when the Timer is edited again.

arendst, 2018-04-19:
You observe this only when going from "Temps" to "Lever.." and back. It's caused by the way the hour select option is changed from 24 hour to 11 hour to accomodate the + and - offset.

It's more a feature than a bug ;-) but I'll look into it anyway.

localhost61, 2018-04-19:
Yes, I confirm that the field is not updated if previous screen shows Hour=00.

But here, it's the case because Timer1 and Timer2 are Suntime related and as I don't use other timers (out of Timer3 for testing the rules behavior) , they have default hour.

Hi Theo,
When I register Timer3 with a non null Time field, I'm directed to Configuration Menu, and when I come back on Timer Configure page for checking, it defaults on Timer1 that is setup with Sunset time and the hour field is effectively 00. Then I select Timer3 and as you stated the Hour field remains unchanged. 

So here, I'm **not** *going from "Temps" to "Lever.."* , I'm  *switching from another Timer view*.
 
I confirm that if I edit Timer1 in standard mode with Hour != 00, then when I register the changes and come back, Timer1 and 3 show the correct hour. I could put Timer 3 in pole position, but I prefer to keep timers 1&2 related to Output 1&2 for my Dual R2, it's easier to remember, and I have memorized that Button 1 is for Down at Sunset, Button 2 for Up at Sunrise.

arendst, 2018-04-20:
Should be solved now.

localhost61, 2018-04-20:
> Should be solved now.
Yes, I have build development #64 and everything works as attended. Thanks.

OfferLa, 2019-08-22:
As a newbie to this great world of Tasmota,
I can confirm the in my place (with NPT active in my network), right after I updated the Latitude and Longitude of my place - Sunrise and Sunset timers worked like a charm.
Thank you so much for this great work!