arendst/Tasmota #2462: 5.12.0l Web Interface Issues

dennyfmn, 2018-04-16:
Hardware (2) Sonoff TH with Si7021 sensor, also Sonoff Basic with added DS18B20 sensor. All three units are acting in a similar fashion.

Systems on my network that are connected via Wi-Fi can connect to all Sonoffs fine. Systems that are connected to the network with hard-wired interfaces have trouble connecting and updating the web pages on all the Sonoffs. It seems worse for systems that are connected to Gigabit switches.

Pages from the Sonoffs only partially load, sometimes without formatting. Sometimes snippets of html are displayed.

I have disabled as many functions as possible in the user_config.h to make the image as small as possible. Smaller images seem to work better than ones that a little larger. There is only 7-8kb of memory free on the Information page. If I do a software reset, they seem to work better for a while.

I know this is pretty vague, but I hope someone will have some insight about this.

Result of Status 0

stat/sonoff06/STATUS {"Status":{"Module":4,"FriendlyName":["Sonoff06"],"Topic":"sonoff06","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
stat/sonoff06/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.3:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"1T01:05:50","StartupUTC":"2018-04-15T21:15:17","Sleep":0,"BootCount":12,"SaveCount":38,"SaveAddress":"FA000"}}
stat/sonoff06/STATUS2 {"StatusFWR":{"Version":"5.12.0l","BuildDateTime":"2018-04-15T16:08:37","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
stat/sonoff06/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":1,"LogHost":"192.168.1.29","LogPort":514,"SSId":["microwap2",""],"TelePeriod":60,"SetOption":["00008129","55818000"]}}
stat/sonoff06/STATUS4 {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
stat/sonoff06/STATUS5 {"StatusNET":{"Hostname":"sonoff06-2920","IPAddress":"192.168.1.104","Gateway":"192.168.1.240","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.240","Mac":"2C:3A:E8:4F:4B:68","Webserver":2,"WifiConfig":3}}
stat/sonoff06/STATUS6 {"StatusMQT":{"MqttHost":"piserver","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4F4B68","MqttUser":"DVES_USER","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff06/STATUS7 {"StatusTIM":{"UTC":"Mon Apr 16 22:21:07 2018","Local":"Mon Apr 16 17:21:07 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"17:38","Sunset":"31:10"}}
stat/sonoff06/STATUS10 {"StatusSNS":{"Time":"2018-04-16T17:21:07","SI7021":{"Temperature":75.4,"Humidity":41.8},"TempUnit":"F"}}
stat/sonoff06/STATUS11 {"StatusSTS":{"Time":"2018-04-16T17:21:07","Uptime":"1T01:05:50","Vcc":3.504,"POWER":"OFF","Wifi":{"AP":1,"SSId":"microwap2","RSSI":100,"APMac":"10:C3:7B:D1:FE:D0"}}}

Thanks!

Frogmore42, 2018-04-17:
"Core":"2_4_1" is known to be bad.  It might be best to start with one of the pre-built binaries that use an older version of the core.

arendst, 2018-04-17:
What option did you set for lwIP Variant in your Arduino IDE? Try if "V2 Lower Memory" works.

You can always install a second copy of the Arduino IDE with 2.4.0. I have dozens of them which makes testing verious version of esp/arduino library a breeze. See wiki what IDE configs are supported although 2.4.1 is known to be a problem.

dennyfmn, 2018-04-17:
It was set to V1.4 Higher Bandwidth. Setting it to V2 Lower Memory and recompiling seems to have helped on the first unit I tested, sonoff06. I’ll update the other two to see if that resolves the issue.

 

Sonoff06 now has 20kb of free memory

Sonoff05 has 7kb of free memory (not updated yet)

 

Thanks for the tip!

dennyfmn, 2018-04-18:
Just to follow up, all three Sonoffs are now working properly on the network and show 20kb of free memory.

V2 Lower Memory did the trick; thanks for your help!

wasi54, 2018-04-30:
@arendst  Just for Info: With core 2.4.0 i never was able to reach the website of any device from a different network via VPN since 12.5.0 ! 
If i compile with Core 2.3.0, everything works fine, also with
Program Version: 5.13.0 
Build Date & Time: 2018-04-30T21:46:47
Core/SDK Version: 2_3_0/1.5.3(aec24ac9)

wasi54, 2018-05-01:
Program Version: 5.13.0a
Build Date & Time: 2018-05-01T20:41:50  (built with atom and platformio on linux)
Core/SDK Version: 2_4_1/2.2.1(cfd48f3)

@arendst Works now with a charme also over VPN, thanks so mutch!

closed 2018-06-02 by ascillato