arendst/Tasmota #2463: Alexa "...... isn't responding", tried everythign i can think of

bryanwarden, 2018-04-16:
Using a Sonoff Basic

Below is response from the console requesting status 0(removed my wifi name, not that it really matters)

00:05:54 CMD: status 0
00:05:54 MQT: stat/diningroom1/STATUS = {"Status":{"Module":1,"FriendlyName":"Dining Room 1","Topic":"diningroom1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:05:54 MQT: stat/diningroom1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":32,"Sleep":0,"BootCount":37,"SaveCount":530,"SaveAddress":"F6000"}}
00:05:54 MQT: stat/diningroom1/STATUS2 = {"StatusFWR":{"Version":"5.11.1g","BuildDateTime":"2018-01-30T17:57:29","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:05:54 MQT: stat/diningroom1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"--------","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
00:05:54 MQT: stat/diningroom1/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":416,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:05:54 MQT: stat/diningroom1/STATUS5 = {"StatusNET":{"Hostname":"diningroom1-7327","IPAddress":"192.168.1.124","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:74:7C:9F","Webserver":2,"WifiConfig":3}}
00:05:54 MQT: stat/diningroom1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.111","MqttPort":1883,"MqttClientMask":"DVES_X","MqttClient":"DVES_747C9F","MqttUser":"mqtt1","MAX_PACKET_SIZE":1024,"KEEPALIVE":15}}
00:05:54 MQT: stat/diningroom1/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 16 23:05:54 2018","Local":"Tue Apr 17 00:05:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
00:05:54 MQT: stat/diningroom1/STATUS10 = {"StatusSNS":{"Time":"2018-04-17T00:05:54","Switch1":"OFF"}}
00:05:54 MQT: stat/diningroom1/STATUS11 = {"StatusSTS":{"Time":"2018-04-17T00:05:54","Uptime":32,"Vcc":3.424,"POWER":"OFF","Wifi":{"AP":1,"SSId":"--------","RSSI":100,"APMac":"C0:56:27:07:10:FE"}}}

So full story, I created a smart fireplace that is controllable with an esp8266 and Alexa.  after that i found these sonoffs and tried a couple out, 2 of them to be exact.  flashed Tasmota, installed, everything seemed fine, i ordered 10 more.  installed all of those into different light switches around the house to control the main lights in my house.  Slowly but surely Alexa became less and less cooperative.  at first she'd occasionally not want to obey, saying whatever light is unresponsive.  I'd ask her again and she'd do it.  this didn't matter if it was a group of lights or one single light, and while this is occurring i'd still have better luck with my esp8266 fireplace, though it would fail half as often as my sonoffs.  

I got frustrated and thought it must be the wemo emulation that was clogging things up so i switched everything to mqtt, got a raspberry pi 3 b+ operating last week, running node-red, and i can control the lights just as smoothly and quickly through node-red on my pc running as i could at the actual light switch, it's instant.  The mqtt works 100% flawlessly, but my Alexa is still unresponsive, and worse off now than it ever was, in most cases she may not ever get any light's on, claiming unresponsive, which is a total load of rubbish as i'm clicking them all day long on mqtt on my pc and also on my phone's mqtt buddy app.

I'd heard something about Ecobee 3's built in Alexa could be causing issues so not only did i try turning that off, i completely disconnected it from the internet, rebooted everything, twice, and still mostly unresponsive.

I have a linksys EA8500, it's by far the happiest I've ever been with a router.  doesn't skip a beat and i never have to reset it(not including me trying to troubleshoot Alexa control.)  I'm by far no expert in WiFi setup, the most I've done really is IP Reservation to keep them static after re-connections so i don't have to chase down the new local IP's and have some control over prioritizing my phone and computers over switches and visitors devices.

What should i try and address.  I've tried updating tasmota but it seems to have updated itself, all of them were up to date.  I've tried resetting the settings on tasmota(even though i don't like the switch settings currently on them) they still do not work correctly.  I've tried mqttretry, read a post about that, but that didn't work.  I've even tried setting up a second(crappy) router to handle just the sonoffs, it was a huge pain, and that didn't work.  I feel i'm missing something, and I apologize if the answer is something simple or already posted but I've searched high and low and cannot find an answer.

Frogmore42, 2018-04-17:
So you turned off emulation on each of the Sonoffs?
If so, how are you expecting Alexa to control them?
Did you install HA-Bridge?

Also, you are using ,"Core":"2_4_0", some people have not had good luck with it and use 2.3.0 instead.

I have lots of things running tasmota and have some of them configured in HA-Bridge so Alexa can control them.  She usually manages to do it correctly.

bryanwarden, 2018-04-17:
Currently i am using mqtt and alexa is running commands through node-red.  as far as what "core" i'm using, i'm not sure, getting these set up, i just followed instructions i found on youtube so i'm not sure, and this was in december i installed all of these.  how would i go about checking(excuse my ignorance)?  I've not tried home assistant, for whatever reason, i thought it was associated with hue bulbs so i never googled it.  I've got an extra sd card, i can install home assistant on my raspberry pi and see if it does the trick.

Frogmore42, 2018-04-17:
As far as I know, Alexa doesn't speak NodeRED or MQTT.  She does know Phillips HUE and some others (and Klingon 😃).

This means you need something on your network that she can understand.  I use this
https://github.com/bwssytems/ha-bridge
It was not easy to setup, since the UI is less than intuitive.  But, once I set t up, it has been great.

You have to provide setup information in HA-Bridge for each device you want to control.  This provides a hue endpoint for Alexa to find and talk to.  HA-Bridge then translates that to MQTT so Tamotsu can understand what to do.

Or, you can enable hue emulation on tasmota and Alexa can talk directly, but many people have had issues with that due to their network configuration.

bryanwarden, 2018-04-17:
I tried all night to install home assistant and the ha bridge.  i could get home assistant up but kept running into errors with bridge.  I ordered a raspberry pi 3 B instead of the B+ so i can just install hass.io which looks to be braindead simple to get that running.  should be able to test things out by friday, i'll see how things go with that.

Jason2866, 2018-04-17:
May you take a look at OpenHab. I dont like Hass.io. -> mqtt command retaining....
"openHAB Hue Emulation Service"
Hue Emulation exposes openHAB items as Hue devices to other Hue HTTP API compatible applications like an Amazon Echo.

Features:
UPNP automatic discovery
Support ON/OFF and Percent/Decimal item types
Can expose any type of item, not just lights
Pairing (security) can be enabled/disabled in real time using the configuration service (under services in the PaperUI for example)
With that i controll my old RF433 power outlet switches via OpenHab with ALEXA
Just google for openhab. There is a RPi3 sd card image. Just google for openhabianpi