arendst/Tasmota #2466: Sonoff POW not reporting energy usage

cloggedDrain, 2018-04-17:
I've flashed a Sonoff POW with versions 5.12.0a (according to sonoff.ino), but it does not report any energy used.  I can verify the energy usage with a Killawat, but the Sonoff is reporting 0V, 0A, 0W etc..

- [ ] Provide the output of command ``status 0`` -
`21:46:46 CMD: status 0
21:46:46 MQT: stat/fishtank_heater/STATUS = {"Status":{"Module":6,"FriendlyName":"Fishtank Heater (Sonoff POW)","Topic":"fishtank_heater","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:46:46 MQT: stat/fishtank_heater/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.158:80/api/arduino/sonoff.ino.bin","Uptime":"0T01:32:07","Sleep":0,"BootCount":9,"SaveCount":30,"SaveAddress":"F6000"}}
21:46:46 MQT: stat/fishtank_heater/STATUS2 = {"StatusFWR":{"Version":"5.12.0a","BuildDateTime":"2018-04-17T14:41:23","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
21:46:46 MQT: stat/fishtank_heater/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.2.10","LogPort":514,"SSId1":"Nazi Zombies","SSId2":"Nazi Zombies 5G","TelePeriod":300,"SetOption":"00000109"}}
21:46:46 MQT: stat/fishtank_heater/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
21:46:46 MQT: stat/fishtank_heater/STATUS5 = {"StatusNET":{"Hostname":"fishtank_heater-1561","IPAddress":"192.168.2.174","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.158","Mac":"5C:CF:7F:7D:26:19","Webserver":2,"WifiConfig":3}}
21:46:46 MQT: stat/fishtank_heater/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.158","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7D2619","MqttUser":"","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:46:46 MQT: stat/fishtank_heater/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 17 20:46:46 2018","Local":"Tue Apr 17 21:46:46 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
21:46:46 MQT: stat/fishtank_heater/STATUS9 = {"StatusPTH":{"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:46:46 MQT: stat/fishtank_heater/STATUS10 = {"StatusSNS":{"Time":"2018-04-17T21:46:46","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
21:46:46 MQT: stat/fishtank_heater/STATUS11 = {"StatusSTS":{"Time":"2018-04-17T21:46:46","Uptime":"0T01:32:07","POWER":"OFF","Wifi":{"AP":1,"SSId":"Nazi Zombies","RSSI":82,"APMac":"80:2A:A8:87:33:39"}}}`

Frogmore42, 2018-04-17:
The relay is off.  It does not report power when the relay is off, unless you change the configuration, but that is only useful if you have modified to hardware to bypass the relay.

cloggedDrain, 2018-04-18:
Thanks for the reply.  

Even when I've toggled the relay, it still shows 0V 0W 0A.  Here is the output with the relay on

`02:41:12 CMD: status 0
02:41:12 MQT: stat/fishtank_heater/STATUS = {"Status":{"Module":6,"FriendlyName":"Fishtank Heater (Sonoff POW)","Topic":"fishtank_heater","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
02:41:12 MQT: stat/fishtank_heater/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.158:80/api/arduino/sonoff.ino.bin","Uptime":"0T06:26:30","Sleep":0,"BootCount":9,"SaveCount":31,"SaveAddress":"F5000"}}
02:41:12 MQT: stat/fishtank_heater/STATUS2 = {"StatusFWR":{"Version":"5.12.0a","BuildDateTime":"2018-04-17T14:41:23","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
02:41:12 MQT: stat/fishtank_heater/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.2.10","LogPort":514,"SSId1":"Nazi Zombies","SSId2":"Nazi Zombies 5G","TelePeriod":300,"SetOption":"00000109"}}
02:41:12 MQT: stat/fishtank_heater/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
02:41:12 MQT: stat/fishtank_heater/STATUS5 = {"StatusNET":{"Hostname":"fishtank_heater-1561","IPAddress":"192.168.2.174","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.158","Mac":"5C:CF:7F:7D:26:19","Webserver":2,"WifiConfig":3}}
02:41:12 MQT: stat/fishtank_heater/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.158","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7D2619","MqttUser":"","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
02:41:12 MQT: stat/fishtank_heater/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 18 01:41:12 2018","Local":"Wed Apr 18 02:41:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
02:41:12 MQT: stat/fishtank_heater/STATUS9 = {"StatusPTH":{"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
02:41:12 MQT: stat/fishtank_heater/STATUS10 = {"StatusSNS":{"Time":"2018-04-18T02:41:12","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
02:41:12 MQT: stat/fishtank_heater/STATUS11 = {"StatusSTS":{"Time":"2018-04-18T02:41:12","Uptime":"0T06:26:30","POWER":"ON","Wifi":{"AP":1,"SSId":"Nazi Zombies","RSSI":78,"APMac":"80:2A:A8:87:33:39"}}}`

Frogmore42, 2018-04-18:
Do you have a POW or an S31 and which did you select from the UI (looks like you selected S31).  Try the other one and see if that works.

cloggedDrain, 2018-04-18:
Aww, that is exactly what the problem was.  I actually have an S31, not a POW.  Once I selected S31 and toggled the relay, power consumption values are reported correctly.

Thanks for the help.

closed 2018-04-18 by cloggedDrain