arendst/Tasmota #2468: Can't access tasmota webpage anymore

gabryk91, 2018-04-17:
- What hardware you are using
Sonoff basic

After successfull flash and first setup I cannot access the tasmota config page at the ip of the sonoff. 
what can it depend on? Thanks

Frogmore42, 2018-04-17:
Lots of things, you will need to hook up serial to see what it thinks is happening.

localhost61, 2018-04-17:
@gabryk91 does it mean that you can still access the web server but only the main page?
If that's the case, I made the same error to enable User mode. It's a trap !!!!
Once in user more you have no way to login in Admin mode. No console, no OTA, you'll have to install a MQTT server or use physical access with a serial terminal (the one integrated in Arduino IDE will do the trick).
Look at #2179  issue and many previous others. It's "by design" for security issue...

gabryk91, 2018-04-18:
No, I can't see the web server anymore, when I navigate to the ip of the sonoff the browser can't find the page. But in my router dhcp list I can see the sonoff ip