arendst/Tasmota #2471: Wemos d1 mini tasmota ir

hugodelaroza, 2018-04-18:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

hugodelaroza, 2018-04-18:
Hola, I'm new with tasamota and sonoff, I'm usina a wemos d1 mini and I conectes a pir sensor and a dth11 sensor, but I want to be able to send ir commands to my TV and Air conditionair equipement but I can't find any tutorial about how to connect the ir led and how to configure tasmota and home assistant. Can anyone help me please?

hugodelaroza, 2018-04-18:
thanks!, I see the board design, but I can't see how to configure tasmota to use the IR led. Can you show me how?

Frogmore42, 2018-04-18:
In the module configuration there is an option for pins for IR Send and another for Receive.  You have to set that for how you have things wired.

I don't us IR, but I see there are some examples from people who have had issues, so searching for examples might be a good idea.

hugodelaroza, 2018-04-19:
Thankyou very much! I will try to build the circuit and send commands over IR. Do you know any circuit to receive IR?

Frogmore42, 2018-04-19:
I don't use IR, but I am sure there are lots of designs available on the internet.  That is where I would look.

Jason2866, 2018-04-19:
Very easy just search for TSOP 1838. It costs a few cents.
Just connect GND and 3.3V. The signal pin to a GPIO Pin from the ESP
Define it as IR Receive. Thats all!!

hugodelaroza, 2018-04-20:
thanks guys! it's amazibg how people I don't even know help me!!

closed 2018-04-20 by hugodelaroza