arendst/Tasmota #2472: Error build code in plarformio

beho1der, 2018-04-18:
Today I build(git clone https://github.com/arendst/Sonoff-Tasmota.git ) the firmware and get an error:

`lib/ESPAsyncUDP-0.21/src/AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':
lib/ESPAsyncUDP-0.21/src/AsyncUDP.cpp:254:57: error: 'udp_set_multicast_netif_addr' was not declared in this scope
udp_set_multicast_netif_addr(_pcb, multicast_if_addr);
^
lib/ESPAsyncUDP-0.21/src/AsyncUDP.cpp:258:36: error: 'udp_set_multicast_ttl' was not declared in this scope`

Leseratte10, 2018-04-18:
Can you try using build 8e0a25f? The next commit adds a bunch of code for the KNX protocol which does not appear in the commit message - was it even intended for this to be added already?

beho1der, 2018-04-18:
I get master branch (git clone  -b master https://github.com/arendst/Sonoff-Tasmota.git), they build without problem. Development build is broken.

localhost61, 2018-04-18:
@beho1der  It's weird, I could build it [yesterday's commit 107379](https://github.com/arendst/Sonoff-Tasmota/commit/107379276be99fae7a72d9e2a347b345997d042a) under Arduino IDE 1.85 and it works fine. Files timestamp was 2018-17-04 08:05:24.
Check if you didn't comment out a define that shouldn't in user_config.h, it can give that kind of error

beho1der, 2018-04-18:
I test current git (git clone https://github.com/arendst/Sonoff-Tasmota.git). Edit only platformio.ini uncomment env_default = sonoff and BUILD, i get error:

```
Compiling .pioenvs/sonoff/lib530/esp-knx-ip-async-udp-0.4.0.15/esp-knx-ip-send.cpp.o
lib/ESPAsyncUDP-0.21/src/AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':
lib/ESPAsyncUDP-0.21/src/AsyncUDP.cpp:254:57: error: 'udp_set_multicast_netif_addr' was not declared in this scope
udp_set_multicast_netif_addr(_pcb, multicast_if_addr);
^
lib/ESPAsyncUDP-0.21/src/AsyncUDP.cpp:258:36: error: 'udp_set_multicast_ttl' was not declared in this scope
udp_set_multicast_ttl(_pcb, ttl);
^
*** [.pioenvs/sonoff/lib076/ESPAsyncUDP-0.21/AsyncUDP.cpp.o] Error 1
```

localhost61, 2018-04-18:
OK, found it !
-> Remove the library Sonoff-Tasmota-development\lib\ESPAsyncUDP-0.21, it is currently not needed but still compiled
I admit that from the archive I only extract the \sonoff directory and only check library changes if I get a build error :-/

localhost61, 2018-04-18:
It seems that the makefile dedendencies under Arduino IDE are more efficient that under Platformio. The libraries are not compiled if not called by the source code. I could add both new libs to the project and recompile without issue.
@beho1der , does it work for you now?

beho1der, 2018-04-18:
I remove folder ESPAsyncUDP-0.21 and esp-knx-ip-async-udp-0.4.0.15 and all build success!

ascillato, 2018-04-19:
Hi,

There is no need to fix the development branch.

The issue that you are experiencing is that both libraries now requires version 2.4.0 of the esp8266 board libraries.

Will not compile under 2.3.0 as explained on the libraries readme files:
[esp-knx-ip readme](https://github.com/envy/esp-knx-ip/blob/async-udp/README.md)
[ESPAsyncUDP readme](https://github.com/me-no-dev/ESPAsyncUDP/blob/master/README.md)

You need to update your esp8266 board libraries as explained [here](https://github.com/esp8266/Arduino)

ascillato, 2018-04-19:
If you are using Arduino, just update the esp8266 library from _Program/Libraries/Manage Libraries_ and look for _esp8266_ and click on update

If you are using Platform.io, just change on the file _platformio.ini_:

```
[env:sonoff]
;platform = espressif8266@1.5.0  ; v2.3.0
platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266
```

or under the language you prefer.

beho1der, 2018-04-19:
It works on platform = espressif8266@1.6.0 Thanks all for help.

ascillato, 2018-04-19:
Great! :+1: 

I made the PR #2487 to add to the Sonoff-Tasmota readme this requirement. Thanks.

localhost61, 2018-04-19:
OK, I confirm I was already in 2.40 under Arduino IDE.

beho1der, 2018-04-19:
Need platform = espressif8266@1.6.0

closed 2018-04-19 by beho1der

ascillato, 2018-04-19:
Yes, but just once.

The first time Platform.io see `platform = espressif8266@1.6.0` will download the library.
Then, using the `platform = espressif8266` will make platform.io use the one just downloaded.

Remember that with this type of change, platform.io also needs a REBUILD C/C++

[image]

arendst, 2018-04-19:
I have no time for testing today but am I right that 2.3.0 still works when KNX is disabled?

If not, then we have to change things as I still want 2.3.0 to be available on non-KNX systems.

beho1der, 2018-04-19:
On  2.3.0 all Build with KNX

reloxx13, 2018-04-19:
> I have no time for testing today but am I right that 2.3.0 still works when KNX is disabled?

nope, KNX commented out and still getting errors while compiling with 2.3.0.

Leseratte10, 2018-04-19:
Probably because here -> https://github.com/arendst/Sonoff-Tasmota/blob/8bc64758504d37486485a014fe69d19bd171e597/sonoff/sonoff_post.h#L36 it doesn't check the KNX config flag.

ascillato, 2018-04-19:
Hi, I'm downgrading the esp-knx-ip library to use 2.3.0 so KNX will run on 2.3.0

I will make the PR soon

The fix will be just:

Delete ESPAsyncUDP library
Update ESP-KNX-IP library

No other change will be needed. And everything under ESP8266 v2.3.0

ascillato, 2018-04-19:
PR #2492 to:

- Update ESP-KNX-IP library
- Delete ESPAsyncUDP library that is no longer used by ESP-KNX-IP library

Compiled and Tested under 2.3.0 with and without `#define USE_KNX`