arendst/Tasmota #2473: How to setup after reset?

gabryk91, 2018-04-18:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ x] What hardware you are using
Sonoff basic

I just reset my sonoff because I couldn't access the webserver anymore, but now what I have to do to setup it again? I see that default user_config file set the ip to 0.0.0.0 how to connect to it then? Thanks

reloxx13, 2018-04-18:
read the wiki.

gabryk91, 2018-04-18:
I did, can you please give me a direct link?

Jason2866, 2018-04-18:
Realy? 
https://github.com/arendst/Sonoff-Tasmota/wiki/Initial-Configuration

gabryk91, 2018-04-18:
Ok thanks! I did it, I saw that instructions early but there is one info missing: I had to connect to the sonoff wifi network called sonoff-xxxx, then I managed to access the webserver!

marcozecchini, 2018-04-19:
I am having the same problem! @gabryk91 how did you solve it? I am trying to do what it is specified in the link above but however I can't connect to it! How do you do steps .2 and .3? Do you put, for example, http://sonoff-0.0.0.0/cn or http://0.0.0.0/cn?

marcozecchini, 2018-04-22:
Hi @gabryk91 I've done what you suggested me and I apparently succeeded in the configuration but now when I type the IP address on the browser, it can't connect to it. Do you have any suggestion?

gabryk91, 2018-04-22:
@marcozecchini si, se hai fatto la procedura teoricamente dovresti vedere una rete wifi che si chiama sonoff-xxxx (dove xxxx è un numero), devi collegarti a quella rete, dopodiché dovrebbe aprirsi automaticamente la pagina dopo qualche secondo, o comunque puntando all'ip default. Ovviamente hai bisogno di un portatile o di uno smartphone se il pc fisso non ha wifi.

gabryk91, 2018-04-22:
Si, direi di si, non ho avuto altre difficoltà dopo

closed 2018-06-21 by stale[bot]

DeanField, 2020-05-08:
I really like the idea of this but the instructions are horrendous.