arendst/Tasmota #2475: Question about Rules (or feature request)

krasatos, 2018-04-18:
Hey there, 
I have just started messing with the new feature implemented: Rules.
I must say that it give me great comfort to know that some stuff are done right on the sonoff and dont depend on mqtt or other connection. This makes me able to place safeguards in order to rest assured that i wont have any problematic behavior when else something disconnects or goes wrong (ie. have my irrigation run all night).

To my question: 
I have been placing safeguards for all relays (5 of them) to turn off at several points, example:
`rules on System#Boot do backlog power1 off;power2 off;power3 off;power4 off;power5 off endon 
on Mqtt#Disconnected do backlog power1 off;power2 off;power3 off;power4 off;power5 off endon`

_(might have some typos, im outside and remembering the code by heart )_
_I also understand that there is a maximum amount of rules due to memory restrictions._

1. Is there some way to refactor the code in order to replace the `backlog power1 off;power2 off;power3 off;power4 off;power5 off` with something like `powerALL off` or would that be possible anytime soon?

2. Another behavior i want to implemet is for all the other relays to turn off (like above) when any relay toggles `PowerX#State`. So i guess ill have to do it 5 times, `when 1 toggle, 2,3,4,5 off` - `when 2 toggle 1,3,4,5 off` and so on. Is there some clever way to shorten this command?

I know the rules implementation is still fresh and experimental, but im guessing a way to contribute to it is talking about it and requesting features :)

Thanks again for this wonderful firmware.
K.

arendst, 2018-04-18:
Well the first rule on System#Boot can easily be solved by changing PowerOnState from 3 to 0 setting all outputs off after power on.

The second one is a nice rule wanting solution ;-)

localhost61, 2018-04-18:
For #2,  isn't it the result of the command: SetOption14 1 // global interlock

localhost61, 2018-04-19:
I just wish to add my 2 cents...
Here is a rule I entered in the console of my Dual R2 used to command a window roller shutter ( build 5.12.0m) :
`CMD: rule`
`RSL: RESULT = {"Rule":"OFF","Once":"OFF","Rules":""}   //oups I forgot to enable rules in user_config.h`
`CMD: rule 1`
`RSL: RESULT = {"Rule":"ON","Once":"OFF","Rules":""}`
`CMD: rule on Rules#Timer=2 do backlog power2 ON;delay 20;power2 OFF;delay 1;power1 ON;delay 1;power1 OFF  endon`
`RSL: RESULT = {"Rule":"ON","Once":"OFF","Rules":"on Rules#Timer=2 do backlog power2 ON;delay 20;power2 OFF;delay 1;power1 ON;delay 1;power1 OFF  endon"}`

_Explanation:_ 
20" is the time it takes to fully open the shutter (Relay2), once up, I rotate the motor backward (Relay1) in order to release the tension because in summer, when it's hot, the rubber joins of the guide rails become very soft and stick to the end slat, when in forced position. At night, when one want to shut them down, they are often locked and one have to turn the motor back and forth to unlock it. It's the reason why, last night I could see on the web console that the sunset timer triggered and the commands occurred, but the shutter was still open. :-[
I'll see next evening if this workaround does the trick.

[image]

arendst, 2018-04-19:
Rules#timer is not timer2!

Rules#timer is a countdown timer useable by rules only.
In your case you should trigger on Clock#timer=2

krasatos, 2018-04-19:
"Well the first rule on System#Boot can easily be solved by changing PowerOnState from 3 to 0 setting all outputs off after power on." @arendst 
Infortunately, due to my (probably ill-configured) HA based mosquito broker, poweronstate does not work for me, so `on Mqtt#Connected do` gives me a peace of mind :). Just want to minimize my code characters.

@localhost61 Great input, i want to set up my garage door to open/close, so my application will be similar. thanks

localhost61, 2018-04-19:
@arendst, thanks for the input. It surrely the reason why the timer2 didn't trigger this morning on sunrise event ;-).
Just for my understanding, isn't it weird that when applying a rule to a timer, like in my previous screenshot, the Output field has to be defined when it's not necessary involved in the rule (In my case both outputs are used)?

arendst, 2018-04-20:
@localhost61 regarding the Output field when Rule is selected: It would need another enable/disable of a HTML item leading to more code while it's just a cosmetic problem.

It would be great if everyone could just use the command line options and get rid of the webserver freeing about 60k for more useful options...

localhost61, 2018-04-21:
Yes the CLI is really powerful, but it's a step ahead from the GUI in the learning curve.

roblad, 2018-04-23:
Hi,

Is it possible to add rule on <trigger>  Serialreceived or SSerialreceived with the received serial <value> %value% for usage as in example Rule

on button1#state do publish cmnd/ring2/power %value% endon on button2#state do publish cmnd/strip1/power %value% endon ? 

Maybe I do not know haw the Event#Anyname when command Event Anyname is executed works, maybe there is already prepared.

closed 2018-06-25 by stale[bot]