arendst/Tasmota #2476: Brennenstuhl RCS 1000N with Sonoff RF Bridge

sroettgermann, 2018-04-18:
HI there,

I'm currently testing the Sonoff RF Bridge flashed with the latest Tasmota release.
I used the webinterface command line to learn the codes from the remote device. I'm able to read the code from the remote and save it. But when I send it the wall plug does not recognize it even if it is in near distance.
Is anyone else experiencing this problem too and is willing to help me with this?

Jason2866, 2018-04-18:
I guess RCS 1000N uses "Intertechno Code" the RF Bridge has problems to decode this correct.
So the learned codes are false and are useless if sent..
Copy the output ot the web console here what received, to verify.

sroettgermann, 2018-05-14:
Sorry for the late reply. Didn’t hat the time yet to test it. Hopefully this week I’ll give it a try.

closed 2018-07-10 by ascillato2