arendst/Tasmota #2478: {"Command":"Unknown"}

arnonel, 2018-04-18:
Hardware: Sonoff RF bridge

14:05:31 CMD: RfKey1 4
14:05:31 RSL: Received Topic /RfKey1, Data Size 1, Data 4
14:05:31 RSL: Group 0, Index 1, Command RFKEY, Data 4
14:05:31 MQT: stat/sonoffRF/RESULT = {"Command":"Unknown"}
Make sure these boxes are checked before submitting your issue - Thank you!

No matter what command i issue, I get: {"Command":"Unknown"}. Example above

Jason2866, 2018-04-18:
Have you received a code and are you in learning mode? Your command means save received data
at key position 1

arnonel, 2018-04-18:
Ignore. I chose Sonoff RF, but looks like choosing Sonoff Bridge made it all work. Not very intuitive, but got there in the end.

Jason2866, 2018-04-18:
Ok, Plz close this issue

closed 2018-04-19 by arnonel

snelweg, 2021-01-16:
> Ignore. I chose Sonoff RF, but looks like choosing Sonoff Bridge made it all work. Not very intuitive, but got there in the end.

Thanks! You saved my day. Was struggling and made the same mistake. It sure is not obvious (it is now though)