arendst/Tasmota #2479: Sticky WiFi issue with AP2 (fallback) istead of AP1

fliesskomma, 2018-04-18:
Issue: All my six Tasmota/Sonoffs (mostly) prefer AP2 (Fallback), allthough AP1 is up and running.

**Hardware:** 
(1) Five Sonoff Basic, one Sonoff Pow, flashed with recent tasmota V5.12.0.
(2) AP1, running on OrangePiZero with Mosquitto Broker. Strongest WiFi-signal around.
(3) AP2, (fallback), also running, on Fritz!box, routing WiFi to the internet. Strong WiFi-signal.

With only one AP running, all Sonoffs behave correctly. Toggling switches works well, and powerdata of Sonnoff Pow updates periodically.

But with 2 APs... 

**1. Case**
Both APs up and running first, all Sonoffs then freshly connected to mains.
Expected result: Sonoffs connect to AP1 and don't care fallback AP2
Obtained result: All Sonoffs ignore AP1 and connect to AP2 immediately . On/OffToggling on the Sonnoffs Website works. The serial protocoll below shows, they don't even care about AP1.

**2. Case**
Going into Sonnoffs WiFi menu and changing SSID of AP2, so the Sonoffs can't login there anymore.
Expected result: Sonoffs connect to AP1
Obtained result: Sonoffs try to connect to AP2. After timeout, because of unknown (because changed) SSID, they all connect to AP1. Ref. serial protocoll below.

**3. Case**
Going into Sonnoffs WiFi menu and changing all SSIDs back to fit again, as in 1. Case.
Expected result: Sonoffs connect to AP1 and not to the fallback AP2
Obtained result: Sonoffs stick to AP1 and the unwanted result of 1. Case is gone.
Things seem to be fixed, but they aren't.

**4. Case**
Taking the Sonoffs from mains and reconnecting again, makes things confusing.
Some Sonoffs connect to AP1, others to AP2. I'm not shure about things happening. The results aren't reproduceable very reliable. I also tried reboot by the Sonnoff Web-Menu.

**5. Case**
If shutting down either one of the APs, the Sonoffs recognizes this and connects to the other one. This works as expected. 

Well, I'm confused, because if the Sonoffs connect to the fallback AP, without caring about AP1, where is the reason and where the solution? Since the Sonoffs don't even care AP1, it's not a matter of timeout. -Which I didn't find Arends code anyway.

Maybe anybody has a hint here?
---
**Serial protocol 1. Case:**
Here you see my sonof05 connecting to AP2 right away, without caring AP1:
00:00:00 Project sonoff05 Sonoff (Topic sonoff05, Fallback DVES_24E27F, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 SonicTemple in mode 11N as sonoff05-0639...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff05-0639.local with IP address 192.168.2.114
11:15:43 MQT: Attempting connection...
11:15:48 MQT: Connect failed to 192.168.1.1:1883, rc -2. Retry in 10 sec

**Serial protocol 2. Case:**
Here you see sonoff05 trying to connect to AP2 first. Because the Sonoff tries with SSID "SonicTempl" (=2. Case), but the AP1 expects "SonicTemple", the connection fails with timeout.
Sonoff now takes AP1, with SSID "OPI" (Orange Pi Zero):
00:00:00 Project sonoff05 Sonoff (Topic sonoff05, Fallback DVES_24E27F, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 SonicTempl in mode 11N as sonoff05-0639...
00:00:09 WIF: Connect failed as AP cannot be reached
00:00:09 WIF: Connecting to AP1 OPI in mode 11N as sonoff05-0639...
00:00:17 WIF: Connected
00:00:17 DNS: Initialized
00:00:17 HTP: Web server active on sonoff05-0639.local with IP address 192.168.1.150
00:00:18 MQT: Attempting connection...
00:00:18 MQT: Connected
00:00:18 MQT: tele/sonoff05/LWT = Online (retained)
00:00:18 MQT: cmnd/sonoff05/POWER = 
00:00:18 MQT: tele/sonoff05/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_24E27F","GroupTopic":"sonoffs"}

fliesskomma, 2018-04-18:
@Jason2866 
Thanks for the hint with the wifi-Commands. I saw them once before, but skipped viewing them until now.

Sorry for the following long reply...
Well, the proof of the pudding is in the eating, so I tried those WiFi commands. I guess I have to tilt over my leading security-thought, protecting AP1 by firewall, which is unreachable from outside my home. And that Sonnoffs showup at AP2 in case of AP1-failure, because I can reach AP2 by remote from outside of my home.

Here comes the long version...
Because of your hint, a beer to my left and the mouse on my right, I took the time studying all of those very, very interesting commands. There is (seems to be) far more potential in conjunction with those commands, than I hoped before. Thanks to Arend.

Anyway, as I understand you, you use the Sonoffs in the same way I want to use them. I.E. The Sonoffs must use AP1! If not available and only if not available, then use AP2. But this is what my Sonoffs don't do! The commands you lead me to, are best for controlling the as-is state, but they don't  help for making AP1 persistent to stay AP1and the fallback-AP to stay fallback-AP after coming reboots.

I just started my Sonoff Network again and watched the Sonoff behaviour.
My Network is like this:
<Internet>-<NAT>-<HomeNetwork (Fallback=AP2)>-<SonoffNetwork (AP1)>
There's some filtering/firewall between the networks and a static routing between AP2 and AP1. So my computer sits in AP2 with a networkscanner scanning AP2 and AP1 in one turn.
**This time all three Sonoffs used AP1 right off from boottime.**
Sonoff01: Configured Primary-AP = AP1, Fallback-AP: Not configured
Sonoff05: Configured Primary-AP = AP2, Fallback-AP: AP1  <<<< THIS ONE USES THE WRONG AP
SonoffPow01: Configured Primary-AP = AP1, Fallback-AP: Not configured

Here's the login protocol of Sonoff05:
00:00:00 Project sonoff05 Sonoff (Topic sonoff05, Fallback DVES_24E27F, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 OPI in mode 11N as sonoff05-0639...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff05-0639.local with IP address 192.168.1.150
21:02:18 MQT: Attempting connection...
21:02:18 MQT: Connected

Here I used all WiFi commands AP, AP0,...AP2:
21:05:01 CMD: ap
21:05:01 MQT: stat/sonoff05/RESULT = {"Ap":"2 (OPI)"}
21:05:12 CMD: ap0
21:05:12 MQT: stat/sonoff05/RESULT = {"Ap":"2 (OPI)"}
21:05:23 CMD: ap1
21:05:23 MQT: stat/sonoff05/RESULT = {"Ap":"2 (OPI)"}
21:05:33 CMD: ap2
21:05:33 MQT: stat/sonoff05/RESULT = {"Ap":"2 (OPI)"}

There's noch switching to the other AP. 
Why?

Then this command:
21:10:39 CMD: wificonfig4
21:10:39 MQT: stat/sonoff05/RESULT = {"WifiConfig":"3 (WPSConfig)"}
Well, I entered wificonfig4 (=Disable wifi config but retry other AP without restart).
Sonoff answered with wificonfig3 (Start WPS config for 1 minute and set as current config tool)

Then I tried all possible wificonfigs... See here, all the same result with wificonfig3:
21:15:45 CMD: wificonfig0
21:15:45 MQT: stat/sonoff05/RESULT = {"WifiConfig":"3 (WPSConfig)"}
21:16:10 CMD: wificonfig1
21:16:10 MQT: stat/sonoff05/RESULT = {"WifiConfig":"3 (WPSConfig)"}
21:16:26 CMD: wificonfig2
21:16:26 MQT: stat/sonoff05/RESULT = {"WifiConfig":"3 (WPSConfig)"}
21:16:34 CMD: wificonfig3
21:16:34 MQT: stat/sonoff05/RESULT = {"WifiConfig":"3 (WPSConfig)"}
21:16:40 CMD: wificonfig4
21:16:40 MQT: stat/sonoff05/RESULT = {"WifiConfig":"3 (WPSConfig)"}
21:16:52 CMD: wificonfig5
21:16:52 MQT: stat/sonoff05/RESULT = {"WifiConfig":"3 (WPSConfig)"}

Well, I think I can't toggle used APs by commandline and I also can't toggle them by hard-coded setup, nor can I toggle them by setup via Webpage. I cannot see a way, not to mention a reliable, reproduceable way to toggle used APs. Thats not very encouraging.

Arend, please help.

More hints still appreciated

Frogmore42, 2018-04-19:
My quick read of the code, support.ino, indicates it does not work the way you think it should.  The way it works is that it considers either AP to be okay to use.  One is not preferred over the other.  I use it this way, because I have two SSIDs, but they are far enough apart that no device can see both.  There is a fiber optic connection between the two locations so they can communicate over the local network.

I don't believe it would be easy to implement what you seem to want.  It might be possible to prefer AP1 on start up.  But, when using AP2 checking to see if AP1 is there might be harder, not sure if the underlying code can scan and maintain a connection.

reloxx13, 2018-04-19:
Command should be
`wificonfig 5`
and not
`wificonfig5`

lloydatron, 2021-03-20:
Hi @fliesskomma I hope you are well. 

I have stumbled onto this slightly old thread after some testing on my setup. Did you get to the bottom of making this AP fallback method work? I am running a network topology much like you are/were here.

Regards