arendst/Tasmota #2480: Pulsetime e Feedback !!

Ronniec10, 2018-04-18:
Hello ,

I have Sonoff Basic with Tasmota 5.12.

I inserted my Sonoff in parallel with the button on the wall, which is connected to a civilian relay.

I have configured pulsetime 5 on the Sonoff and it works well with my lighting system,. The problem is:
1) I want to see the correct feedback when I activated my sonoff output (and therefore excited my relay) (ON)
2) I want the feedback, as above, but when I turn off (off)
3) I want the feedback even when I press the button on the wall that is connected only to the house relay.

In this way I guarantee the functioning of my light system even if the sonoff should break. But I do not have the feedback of the light on because when I press ON on my sonoff and the electrical impulse occurs then the Sonoff is set to OFF !! and viceversa.

Please help me

Jason2866, 2018-04-18:
No offend! Read the wiki. Think and read the wiki again. 
You wont get a working solution if you dont understand what you are doing.
If you dont like this or dont have the time or energy, Tasmota is not the solution for you.

Ronniec10, 2018-04-18:
Of course you were offending me! If I asked it 'cause I do not know otherwise there is no reason to create this forum! Who are you? If you can not help me or if you can not help me, why do you speak ?! Who asked you anything? Or if today for you it's a bad day, unleash your frustration with another person and do not allow yourself!

Jason2866, 2018-04-18:
No this is not a forum. This is github issue. This is normally the place where errors in the software are posted...

Ronniec10, 2018-04-18:
are you kidding me? but if you browse the questions on the wiki you can see that there are the most disparate questions !!! but anyway, I did not understand who you would be. get out of the way! you're not giving me any input. so go your way and do not break the balls !!

Frogmore42, 2018-04-18:
Many people use this a forum, but it is not really a forum.  I get that you are frustrated and can't figure out what is going on.  But, you have not provided any actionable information on a problem.  The suggestion to read the wiki might seem offensive, but it is probably the best way to understand how Tasmota works.  That and searching the issues list for other reported issues that are similar to what you are trying to do.  Also, what have you tried to solve your problem and what results did you get?  Have you looked at the log to see what it is showing.  There really is a lot of information in the wiki and the log, though some of it is hard to find.

curzon01, 2018-04-19:
@Ronniec10 - the simple answer is your requirement can not be fullfilled with a Sonoff Basic. You use the Sonoff as a toggle and there is simply no technical feedback info from your light to your Sonoff Basic. It might work using a Sonoff Pow measure if power is consuming (light is on) or not (light is off).

Ronniec10, 2018-04-19:
I have tried everything. I've read the wiki several times. Currently I use pulsetime so when I send the command the sonoff gets excited and then it de-energizes and up to this everything is ok, in this mode it works well with my relay. Unfortunately, however, I do not have the feedback. According to you (at least you talk to help and not just to say nonsense) can I use the GP14 to get a light feedback on? Instead say 'read the wiki give me a way to follow if you know the software well or shut up and do not talk anymore'!

Ronniec10, 2018-04-19:
Hi Curzon, my first message was not related to you. How do I turn my basic sonoff with tasmota firmware into sonoff pow? which sensor should I buy to read the voltage present? And above all, how should I modify my configuration.yaml file in order to read the data in Homeassist? (just in case anyone knows how to do with Homeassist, I know it's not the right place to talk about Home Assist)

Frogmore42, 2018-04-19:
@Ronniec10 
While your approach might serve you well in some places, it is probably not the best way to get help.
While you might think your question is clear, I don't believe anyone else understands exactly what you are asking.

You can't turn a Sonof Basic into a POW.  They are completely different products.  That is covered in the wiki.

If you are asking how a Sonof can tell if your light is on when you have turned it on by some other means, it can't.  That was the question I think @curzon01 was answering.  If you are asking how to tell when Tasmota changes the status of the light, which might have been controlled by a switch connected to the Sonoffs GPIO, there is a MQTT response for that.

Ronniec10, 2018-04-19:
can not it be turned into a pow? I currently turned a basic sonoff into a Th 10. I connected a Dht11 (temperature) on Gpio 14 and it works great. I thought you could do the same thing on Pow, but I do not know how the voltage reading sensor works

Frogmore42, 2018-04-19:
The POW is very different and dangerous, so you can't add sensors to it.  It sounds like you have the switch on you wall in parallel with a Sonoff.  I don't know where you live, but I am pretty sure that would be a violation of the National Electric Code in the USA.  I also still don't know exactly what you are trying to achieve.  I am using a POW to measure and control a dehumidifier.  The dehumidifier is configured to always run.  It is plugged into the POW.  The POW is controlled by a flow in NodeRed that listens to a NodeMCU running Tasmota that has a humidity sensor (the DHT11 is a very poor quality sensor, so I use SHT15, HTU21, or BME280).  I have a rain gauge measuring the water output from the dehumidifier so I can see everything is working and keep track of the energy use and efficiency.  The default control by the dehumidifier is very inefficient.  So, it is possible to do some interesting things with this.  But, it took me several weekends to get to the point where I understood the basics of how things work.  Maybe you already know, but the questions you are asking make it seem that you don't have a good understanding of how all the pieces fit together.

Ronniec10, 2018-04-19:
and in your opinion .. if I had all the information I need I would have come to this place to ask questions? but by chance are you all in the same room from which you are answering? that is, rather than trying to give a concrete help tirste out arguments that I have not even heard of in a work safety class. Ok. I get it . That's enough . You are not giving me any input. It's just a waste of time. I'll be alone. End of the episode

ascillato, 2018-06-04:
#1420 can be a solution for you?

closed 2018-06-12 by ascillato2