arendst/Tasmota #2481: POW Restart

Lu-Fi, 2018-04-18:
Hi, i have multiple POW / Basic Switches here.
If the devices are restarting all of them having a short switch cycle (off/on).
Is it possible to prevent this ? 
If for example WIFI is not available the devices will restart every Minute.
And some Devices Powered by using this Switches don't like this.

Best Regards

LuFi

Jason2866, 2018-04-18:
Hass.io?  A log file and status 0 could help.

lobocobra, 2018-04-18:
Hi
I had similar issues with an older version of the firmware. Are you at the latest level? 
=> Did you check your config settings?
=> Do you use MQTT and do you retain the state?

Did you read the meaning of all commands in the wiki? It took me 3 month to understand, that even I have to read the wiki, in order to put the right settings (guessing did not help me).

In my case the error was sitting in front of the SONOFF and the one issue with Pulsetimer I found was fixed since.
=> Trust the Wiki :-)

Lu-Fi, 2018-04-19:
I Think this could explain also my Problem.

https://github.com/arendst/Sonoff-Tasmota/issues/1034

Will test this later

closed 2018-04-20 by Lu-Fi