arendst/Tasmota #2482: Support I2C slave mode

renne, 2018-04-18:
- [x] What hardware you are using
Ai Thinker ESP8266-12F

The WiFi of the ESP8266 is great if you don't want to install cables. Like all radio systems it is not tamper-free and can be disabled by jammers e.g. sending disconnect packets. For security appliances good old cables are necessary. On my search for an easy to install bus system I realized only differential buses are reliable and I wanted to use a CAN bus. The drawback: CAN controllers usually use SPI which costs a lot of GPIOs on the ESP8266. So why not use the I2C as a long range bus system? Just connect a CAN-transceiver to SDA and SCL on the ESP8266 and I2C can handle long distances e.g. in a house.

As there can be only one master on an I2C bus I suggest to support I2C slave mode in Tasmota. That way Tasmota can use WiFi for non-critical high-volume data (e.g. environment sensors) and differential I2C for critical low-volume data (e.g. door/window contacts, motion detectors, glass breakage detectors, smoke detectors, gas detectors, ...).

[Expressif forum discussion](https://bbs.espressif.com/viewtopic.php?f=49&t=2092)
[esp8266/Arduino Support for I2C slave mode issue](https://github.com/esp8266/Arduino/issues/1330)
[esp8266/Arduino repository pull request](https://github.com/esp8266/Arduino/pull/3237)

lobocobra, 2018-04-18:
Cool idea, let me know, when you programmed it.
=> I will test it :-)

renne, 2018-04-19:
2 bipolar buffers to demux RX/TX of SCL and SDA and 4 CAN-transceivers cost about 6,- EUR. An ESP32 + CAN-transceiver is just 3,50 EUR. It's not really economic unless someone knows cheap bipolar buffers for RX/TX demuxing. :-(