arendst/Tasmota #2484: Name of 'Backup Configuration' file truncated if Friendly Name contains 'spaces'

dweston, 2018-04-18:
If the Friendly Name contains a space character, when an attempt is made to save a Backup Configuration file the auto created file name is truncated. See following screen excerpt.

[image]

- [x] What hardware you are using
Sonoff TH16
Source Code version 5.12.0m

- [x] Provide the output of command ``status 0``
`20:10:52 CMD: status 0
20:10:52 MQT: stat/sonoff-winefan/STATUS = {"Status":{"Module":4,"FriendlyName":["Wine Fan"],"Topic":"sonoff-winefan","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:10:52 MQT: stat/sonoff-winefan/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:08","StartupUTC":"2018-04-18T18:57:44","Sleep":0,"BootCount":7,"SaveCount":13,"SaveAddress":"F7000"}}
20:10:52 MQT: stat/sonoff-winefan/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-19T04:49:12","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
20:10:52 MQT: stat/sonoff-winefan/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vista_TC_1","dionw"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
20:10:52 MQT: stat/sonoff-winefan/STATUS4 = {"StatusMEM":{"ProgramSize":501,"Free":500,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
20:10:52 MQT: stat/sonoff-winefan/STATUS5 = {"StatusNET":{"Hostname":"sonoff-winefan-4318","IPAddress":"10.0.1.23","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"5C:CF:7F:A0:10:DE","Webserver":2,"WifiConfig":3}}
20:10:52 MQT: stat/sonoff-winefan/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.14","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A010DE","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:10:52 MQT: stat/sonoff-winefan/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 18 19:10:52 2018","Local":"Wed Apr 18 20:10:52 2018","StartDST":"Sun Oct 07 02:00:00 2018","EndDST":"Sun Apr 01 03:00:00 2018","Timezone":1,"Sunrise":"21:19","Sunset":"08:28"}}
20:10:52 MQT: stat/sonoff-winefan/STATUS10 = {"StatusSNS":{"Time":"2018-04-18T20:10:52","DS18B20":{"Temperature":19.7},"TempUnit":"C"}}
20:10:52 MQT: stat/sonoff-winefan/STATUS11 = {"StatusSTS":{"Time":"2018-04-18T20:10:52","Uptime":"0T00:13:08","Vcc":3.320,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Vista_TC_1","RSSI":70,"APMac":"70:73:CB:BD:0F:33"}}}`

closed 2018-04-19 by arendst

Leseratte10, 2018-04-20:
Your solution still breaks when the friendly name contains a semicolon (wrong file name). And it breaks in Chrome, if the friendly name contains a comma (download not possible at all). 

Why not just add the quotes in addition to the "space-to-underscore"-patch (and use backspace to escape possible quotes in the friendly name itself)?

arendst, 2018-04-20:
Give it another try. 

This time ALL non-alphanumeric characters are replaced by underscores using a small (important) code footprint.