arendst/Tasmota #2485: After the flashing I can't communicate with Sonoff Basic

marcozecchini, 2018-04-18:
Hi all, I am new with the Sonoff devices. I have a Sonoff Basic and I' ve flashed in it the Tasmota firmware following these configurations:

/*
  user_config.h - user specific configuration for Sonoff-Tasmota
*/

#ifndef _USER_CONFIG_H_
#define _USER_CONFIG_H_

// -- Localization --------------------------------
//#define MY_LANGUAGE            de-DE           // German in Germany
//#define MY_LANGUAGE            en-GB           // English in Great Britain. Enabled by Default
//#define MY_LANGUAGE            es-AR           // Spanish in Argentina
//#define MY_LANGUAGE            fr-FR           // French in France
//#define MY_LANGUAGE            it-IT           // Italian in Italy
//#define MY_LANGUAGE            nl-NL           // Dutch in the Netherlands
//#define MY_LANGUAGE            pl-PL           // Polish in Poland
//#define MY_LANGUAGE            zh-CN           // Chinese (Simplified) in China

// -- Project -------------------------------------
#define PROJECT                "sonoff"          // PROJECT is used as the default topic delimiter and OTA file name
                                                 //   As an IDE restriction it needs to be the same as the main .ino file

#define CFG_HOLDER             1        // [Reset 1] Change this value to load following default configuration parameters
#define SAVE_DATA              1                 // [SaveData] Save changed parameters to Flash (0 = disable, 1 - 3600 seconds)
#define SAVE_STATE             1                 // [SetOption0] Save changed power state to Flash (0 = disable, 1 = enable)

// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.1.1"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.2.27"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "WIFISSID"      // [Ssid1] Wifi SSID
#define STA_PASS1              "PASSWORD"  // [Password1] Wifi password
#define STA_SSID2              "indebuurt2"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "VnsqrtnrsddbrN"  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)

// -- Syslog --------------------------------------
#define SYS_LOG_HOST           "domus1"          // [LogHost] (Linux) syslog host
#define SYS_LOG_PORT           514               // [LogPort] default syslog UDP port
#define SYS_LOG_LEVEL          LOG_LEVEL_NONE    // [SysLog]
#define SERIAL_LOG_LEVEL       LOG_LEVEL_INFO    // [SerialLog]
#define WEB_LOG_LEVEL          LOG_LEVEL_INFO    // [WebLog]

// -- Ota -----------------------------------------
#define OTA_URL                "http://domus1:80/api/arduino/" PROJECT ".ino.bin"  // [OtaUrl]

// -- MQTT ----------------------------------------
#define MQTT_USE               1                 // [SetOption3] Select default MQTT use (0 = Off, 1 = On)
// !!! TLS uses a LOT OF MEMORY (20k) so be careful to enable other options at the same time !!!
//#define USE_MQTT_TLS                             // EXPERIMENTAL Use TLS for MQTT connection (+53k code, +20k mem) - Disable by //
                                                 //   Needs Fingerprint, TLS Port, UserId and Password
#ifdef USE_MQTT_TLS
  #define MQTT_HOST            "m20.cloudmqtt.com"  // [MqttHost]
  #define MQTT_FINGERPRINT     "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  // [MqttFingerprint]
  #define MQTT_PORT            20123                // [MqttPort] MQTT TLS port
  #define MQTT_USER            "cloudmqttuser"      // [MqttUser] Mandatory user
  #define MQTT_PASS            "cloudmqttpassword"  // [MqttPassword] Mandatory password
#else
  #define MQTT_HOST            "domus1"          // [MqttHost]
  #define MQTT_PORT            1883              // [MqttPort] MQTT port (10123 on CloudMQTT)
  #define MQTT_USER            "DVES_USER"       // [MqttUser] Optional user
  #define MQTT_PASS            "DVES_PASS"       // [MqttPassword] Optional password
#endif

#define MQTT_BUTTON_RETAIN     0                 // [ButtonRetain] Button may send retain flag (0 = off, 1 = on)
#define MQTT_POWER_RETAIN      0                 // [PowerRetain] Power status message may send retain flag (0 = off, 1 = on)
[... 118 lines cut ...]
  #define USE_WS2812_CTYPE     1                 // WS2812 Color type (0 - RGB, 1 - GRB, 2 - RGBW, 3 - GRBW)
//  #define USE_WS2812_DMA                         // DMA supports only GPIO03 (= Serial RXD) (+1k mem). When USE_WS2812_DMA is enabled expect Exceptions on Pow

#define USE_ARILUX_RF                            // Add support for Arilux RF remote controller (+0k8 code, 252 iram (non 2.3.0))

/*********************************************************************************************\
 * Compile a minimal version if upgrade memory gets tight ONLY TO BE USED FOR UPGRADE STEP 1!
 *   To be used as step 1 during upgrade.
 *   Step 2 is re-compile with option BE_MINIMAL commented out.
 *   !!! Needed for next release of Arduino/ESP8266 (+22k code, +2k mem) !!!
\*********************************************************************************************/

//#define BE_MINIMAL                               // Minimal version if upgrade memory gets tight (-45k code, -2k mem)

/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

#endif  // _USER_CONFIG_H_

Using Platoformio, the upload goes fine (the device even appears connected to my router)  however I can't communicate with it neither with serial communication nor with the webpage. Am I missing something in the configuration? I am pressing the button for over fourty seconds

reloxx13, 2018-04-19:
Does the LED blink once when pluging in power?
Does the LED blink constantly?
Does the button trigger the relais/led?

marcozecchini, 2018-04-19:
Yes, the LED blink when you plug in and it seems to trigger correctly the relais/led. I' ve tried to press four-short times the button to reconfigure the wifi but I still can't connect to its ip.

curzon01, 2018-04-19:
you can use your flash interface and a serial terminal to observe the Tasmota serial log whats happened even you try to multi-press your button

marcozecchini, 2018-04-19:
@curzon01 The problem is that the serial terminal of the sonoff doesn't send anything or I can't see if it is sending something

curzon01, 2018-04-20:
As you already defined `#define SERIAL_LOG_LEVEL LOG_LEVEL_INFO // [SerialLog]` the serial log sends data when starting the device. If not something is wrong:
- intermediate: check if your terminal program uses 115200 8N1
- if not working try to flash a ready compiled binary from [release repository](https://github.com/arendst/Sonoff-Tasmota/releases) and try display the serial log. This should work as first step before you go next step using your own compilation.

marcozecchini, 2018-04-22:
Thank you for your help @curzon01, I' ve used a different serial terminal (CoolTerm) and it works! 
Now the problem is that even if I've succeeded in the web server configuration it doesn't respond when I type its IP-address in the browser. Which might be the problem?

marcozecchini, 2018-04-22:
Thank you for your help @curzon01, I' ve used a different serial terminal (CoolTerm) and it works! 
Now the problem is that even if I've succeed in the webserver configuration it doesn't respond when I type its IP-address in the browser. Which might be the problem?

marcozecchini, 2018-04-22:
I tested this release but I flashed it using Platformio and downloading the zip. Do you think that this is the problem? Should I flash tasmota using the binary release?

curzon01, 2018-04-22:
You should try the binary first, then will see

closed 2018-06-21 by stale[bot]