arendst/Tasmota #2489: 4 channel relay works opposite

ashishcchauhan, 2018-04-19:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

ashishcchauhan, 2018-04-19:
I Flash tasmota on NodMCU and controlling 4 relay board, Its works but there is little problem.
When I trigger, It works opposite, means when I turning ON my relay goes OFF and When I  turning OFF it goes ON.
So Please help me to solve this problem.

My tasmota settings are below.
Module type is 18.Generic
D6 GPIO12 Relay1      -   Relay 01 
D7 GPIO13 Led1i       -   Relay 02
D5 GPIO14 Sensor     -   Relay 03
D8 GPIO15                  -   Relay 04
It will be very helpful to me if you can solve this problem.
Sorry  for my English.
Thanks,﻿

Jason2866, 2018-04-19:
This is easy to solve. Just use the inverted relay option. Example
Relay1i

ashishcchauhan, 2018-04-19:
hey jason,
 thank for your reply but i am newbie in this project so I not understanding what you suggested can you explain  little more.

Thanks.

Jason2866, 2018-04-19:
Open the web interface of your NodeMCU  (http://....)
Goto configuration -> configuration module ->  choose the Gpio pin with the wrong on off behaviour 
and change it from Relay1 to Relay1i.
Save this configuration  (blue button, bottom of the page)
The i in Relay1i stands for inverted. So the relay acts you want to

ashishcchauhan, 2018-04-19:
Thank you sooooo much jason....its don....thanks again....

closed 2018-06-05 by ascillato

issa13, 2018-07-13:
But how can I open the Web page of my NodeMCU??