arendst/Tasmota #2494: Has anyone tested Tasmota on Sonoff Pro

aaamitsingh, 2018-04-19:
Hi everyone I'm facing issue with Sonoff Pro, with default firmware its working fine but after flashing Tasmota it's behaving in a very slow and odd way. When I press the button 1st time its work slowly but after some time relay on/off happens but after that buttons stop responding even the 1st button.
Again If I reboot the device same things happen. 
Apart from the relay and led on/off nothing happens with the first button.

I have also sonoff 4ch and sonoff sv, both work fine with the Tasmota. Can anyone please confirm me that they are not facing this kind of issue with Tasmota and sonoff pro.

Leseratte10, 2018-04-19:
Sonoff Pro? Are you talking about the Sonoff 4CH Pro? I use two 4CH Pro devices with the newest Tasmota and they work without issues. Did you try re-flashing the device?

reloxx13, 2018-04-19:
if you dont use domoticz, disable it in user_config.h

if you dont use mqtt, disable it on the webui or user_config.h

Alveran81, 2018-04-21:
I have same Problem. Its realy slow. And Chanel 1 had a Magic life. It Turns on and off itself.

Jason2866, 2018-04-24:
@Alveran81 Do you have connected additional switches? If yes, disconnect it for testing.
If problem is gone, you have to install Pullup resistors. If nothing connecting plz post Status 0
and a log when Channel 1 does magic switching

Alveran81, 2018-04-25:
There is no external Switch installed for Channel 1. What is Status 0 ?

Here is the log.

10:33:23 APP: Button1 level 1-0
10:33:23 MQT: stat/Sonoff4chpro2/RESULT = {"POWER1":"ON"}
10:33:23 MQT: stat/Sonoff4chpro2/POWER1 = ON
10:33:23 APP: Button1 level 0-1
10:33:24 CFG: Saved to flash at F6, Count 1297, Bytes 1712
10:36:14 MQT: tele/Sonoff4chpro2/STATE = {"Time":"2018-04-25T10:36:14","Uptime":"5T01:26:29","Vcc":3.427,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Zyxel","RSSI":100,"APMac":"08:96:D7:90:D1:8D"}}
10:38:41 APP: Button1 level 1-0
10:38:41 MQT: stat/Sonoff4chpro2/RESULT = {"POWER1":"OFF"}
10:38:41 MQT: stat/Sonoff4chpro2/POWER1 = OFF
10:38:41 APP: Button1 level 0-1
10:38:42 CFG: Saved to flash at F5, Count 1298, Bytes 1712
10:41:14 MQT: tele/Sonoff4chpro2/STATE = {"Time":"2018-04-25T10:41:14","Uptime":"5T01:31:29","Vcc":3.427,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Zyxel","RSSI":100,"APMac":"08:96:D7:90:D1:8D"}}
10:46:14 MQT: tele/Sonoff4chpro2/STATE = {"Time":"2018-04-25T10:46:14","Uptime":"5T01:36:29","Vcc":3.427,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Zyxel","RSSI":100,"APMac":"08:96:D7:90:D1:8D"}}
10:51:03 APP: Button1 level 1-0
10:51:03 MQT: stat/Sonoff4chpro2/RESULT = {"POWER1":"ON"}
10:51:03 MQT: stat/Sonoff4chpro2/POWER1 = ON
10:51:03 APP: Button1 level 0-1
10:51:04 CFG: Saved to flash at F4, Count 1299, Bytes 1712
10:51:14 MQT: tele/Sonoff4chpro2/STATE = {"Time":"2018-04-25T10:51:14","Uptime":"5T01:41:29","Vcc":3.427,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Zyxel","RSSI":100,"APMac":"08:96:D7:90:D1:8D"}}
10:56:14 MQT: tele/Sonoff4chpro2/STATE = {"Time":"2018-04-25T10:56:14","Uptime":"5T01:46:29","Vcc":3.427,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Zyxel","RSSI":100,"APMac":"08:96:D7:90:D1:8D"}}
10:59:53 APP: Button1 level 1-0
10:59:53 MQT: stat/Sonoff4chpro2/RESULT = {"POWER1":"OFF"}
10:59:53 MQT: stat/Sonoff4chpro2/POWER1 = OFF
10:59:53 APP: Button1 level 0-1
10:59:54 CFG: Saved to flash at FB, Count 1300, Bytes 1712
11:00:46 APP: Button1 level 1-0
11:00:46 MQT: stat/Sonoff4chpro2/RESULT = {"POWER1":"ON"}
11:00:46 MQT: stat/Sonoff4chpro2/POWER1 = ON
11:00:46 APP: Button1 level 0-1

titomontes, 2018-04-30:
If you are not using MQTT the disable it. It´s a known issue: If no MQTT server is found, the device will be looping to find one, hence the slugishness.

Status 0 is a console command to display the status of the sonoff

Alveran81, 2018-06-07:
Sorry. Now two chanels doing this. 
Where can i found arduino ide Version?

Jason2866, 2018-06-07:
Upgrade with the precompiled version from release path.
The latest version uses (again) arduino ide 2.3.0.
Your settings will be kept.

Jason2866, 2018-06-08:
Reset the device. Maybe a bad config that is stored. If that not solve the issue, the device hardware is buggy...

Alveran81, 2018-06-09:
I resetet Configuration. It works for about 12 hours. And then again same Problem.
I did not change anything.

Jason2866, 2018-06-09:
I think the device has a hardware defect.

closed 2018-07-10 by ascillato2