arendst/Tasmota #2495: Switch multiple relays running tasmota on Wemos D1 mini

MPJK77, 2018-04-19:
Hello Everyone,

I have a 4 relay module I want to control with tasmota 5.12 on a Wemos D1 mini.
For now I am able to switch one relay using D5 (GPIO14 switch) and D6 (GPIO12 relais)
But there are (minimal) two relays more I need to switch.
Is there any way to achieve this with only one wemos d1 running Tasmota?

The end goal is to automate my sunscreen using Hassio.
I have cloned a Somfy Telis 4 remote control for my somfy RTS motor.
Now each relay needs to operate a button.
Relay 1 to open the sunscreen
Relay 2 to close the sunscreen
Relay 3 to stop the sunscreen
Relay 4 to do something else if possible.

Thanks in Advance!

reloxx13, 2018-04-19:
if you set the device as generic, you can set relaises to gpios you want.

if its not enough, check wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

Jason2866, 2018-04-19:
Take a look here
https://escapequotes.net/esp8266-wemos-d1-mini-pins-and-diagram/

reloxx13, 2018-04-19:
they are not connected, you need to match them yourself

e.g. 
gpio 14 switch1 and gpio12 relais1  = gpio14 handles gpio12
gpio 02 switch2 and gpio03 relais2 = gpio02 switches gpio03

the number after the type (switch/relais  etc.) is the mapping.

MPJK77, 2018-04-19:
Thank you very much!

closed 2018-06-05 by ascillato