arendst/Tasmota #2496: Add Support for the SI1145 UV Index Sensor

FabianHM, 2018-04-19:
Please add Support for SI1145 UV Index Sensor.
This board is the perfect complement to the already supported VEML6070 UV sensor.

More Information and the Library of this Board you can find here (Adafruit):
https://learn.adafruit.com/adafruit-si1145-breakout-board-uv-ir-visible-sensor/downloads

Thanks
Fabian

FabianHM, 2018-05-08:
For all who also interested in the UV index I wanted to report that there is this great sensor for € 3.50.
This sensor is a useful addition to the already supported VEML6070 UV level sensor.

@arendst : I would be very happy if this sensor were also supported by Tasmota.
If further parameters are needed please message.
Fabian

arendst, 2018-05-08:
Now that I'm flooded with donations ;;--)) I finally can order some of these and start fiddling around with them.

Stay tuned.

FabianHM, 2018-05-10:
My sensor is ready and waiting for Tasmota, I hope your sensor will arrive soon :-)
have a look at your paypal account ;-)

arendst, 2018-05-12:
Give it a try.

FabianHM, 2018-05-13:
@arendst: Unfortunately I can not test the next days. As soon as I'm back I'm looking forward to the test 👍

FabianHM, 2018-05-22:
Today I finally wanted to test my SI1145 UV Index Sensor. Unfortunately, the part after 3 sec. (at 3.3 volts) so hot that it is good for more than heating instead of a sensor.
I'll probably order one more and unfortunately have to wait another 5 weeks :-(

ascillato2, 2018-07-21:
Hi,

Feature added to Tasmota. Closing issue. Thanks

closed 2018-07-21 by ascillato2