arendst/Tasmota #2497: cannot flash, strange output

kbaras, 2018-04-19:
Hi! 

I flashed tasmota to sonoff DUAL. tried to use two switches on button 0 and 1 with limited success. It worked for about 10minutes, but now I experience strange behaviour. It does not react anymore to button push. Sometimes the leds are on, but sometimes it seems that the device is dead.

It has the following output:
00:00:24 MQT: Connect failed to 192.168.1.226:1883, rc -2. Retry in 10 sec
00:00:29 UPP: Multicast (re)joined
00:01:21 MQT: Attempting connection...
00:01:26 MQT: Connect failed to 192.168.1.226:1883, rc -2. Retry in 10 sec
00:01:30 UPP: Multicast (re)joined
00:02:03 RSL: ��������������������������������������������������������������������������������������������������� = {"Time":"1970-01-01T00:02:03","Upt
ime":1}
00:02:22 MQT: Attempting connection...
00:02:27 MQT: Connect failed to 192.168.1.226:1883, rc -2. Retry in 10 sec
00:02:31 UPP: Multicast (re)joined
00:03:23 MQT: Attempting connection...
00:03:28 MQT: Connect failed to 192.168.1.226:1883, rc -2. Retry in 10 sec
00:03:32 UPP: Multicast (re)joined

Is it possible that I have damaged the device? If so, how??? I only closed button0/1 with gnd.

Frogmore42, 2018-04-20:
Make sure these boxes are checked before submitting your issue - Thank you!
 What hardware you are using (there are two versions of the Dual)
 Provide the output of command status 0
 
You device does not have the correct time and it seems to be having difficulty connecting to the MQTT.  This probably means it is having difficulty using the network.  If you built the FW, it is possible you have (automatically) selected a bad configuration of components (status 0 would help show if that is the problem, which is why the issue has that as necessary to answer)

kbaras, 2018-04-20:
Hi!

Ok, done!
This is a sonoff dual R2.

Status 0 output is:
_00:17:11 CMD: status 0
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoffdual1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff-nappali:80/api/arduino/sonoffdual1.ino.bin","Uptime":1,"Sleep":255,"BootCount":62,"SaveCount":267,"SaveAddress":"F9000"}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusFWR":{"Version":"5.11.1i","BuildDateTime":"2018-04-12T23:00:27","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"mosquitto","LogPort":514,"SSId1":"HOME_WIRELESS","SSId2":"HOME_WIRELESS-sagem","TelePeriod":300,"SetOption":"00000009"}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusNET":{"Hostname":"sonoffdual1","IPAddress":"255.255.255.255","Gateway":"255.255.255.255","Subnetmask":"255.255.255.255","DNSServer":"255.255.255.255","Mac":"DC:4F:22:80:40:21","Webserver":2,"WifiConfig":3}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusMQT":{"MqttHost":"192.168.1.226","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_804021","MqttUser":"kbaras","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusTIM":{"UTC":"Thu Jan 01 00:17:11 1970","Local":"Thu Jan 01 00:17:11 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusSNS":{"Time":"1970-01-01T00:17:11"}}
00:17:11 RSL: ˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙˙ = {"StatusSTS":{"Time":"1970-01-01T00:17:11","Uptime":1,"Vcc":3.342,"POWER":"ON","Wifi":{"AP":1,"SSId":"HOME_WIRELESS","RSSI":100,"APMac":"10:BF:48:3D:13:EC"}}}_

Strange is that it worked at first, it was able to connect to the wifi, and to mosquitto (and I was able to control it via mqtt). I haven't changed anything in config.

kbaras, 2018-04-20:
Additional question: how can I get to the admin page if I don't know the ip of the device (e.g. in this case)? Or is it possible to change the config via serial commands - without a reflash?

reloxx13, 2018-04-21:
`{"Hostname":"sonoffdual1","IPAddress":"255.255.255.255","Gateway":"255.255.255.255","Subnetmask":"255.255.255.255","DNSServer":"255.255.255.255","Mac":"DC:4F:22:80:40:21","Webserver":2,"WifiConfig":3}}`

dude, get your things up. if you dont even understand what an ip is stop changing it and read theory first.

revert everything you made in the user_config.h, change the cfg_holder -1, flash again, change cfg_holder again +1 and flash again to get a clean and working device.

and again read, read and read. read the status 0, read and go though your router web ui, read the wiki, read about network ip addresses and so on.

you didnt even read the draft you get when openening an issue.

localhost61, 2018-04-21:
@reloxx13 It seems normal to me if the device is waiting an IP from DHCP.

Frogmore42, 2018-04-21:
Well something is really not right with networking.  This likely means you either have misconfigured the software or your Wi-Fi network is not well configured.

Some people have had difficulty with core 2.4.0, but it has worked for others.

How did you get the log?  The voltage listed for VCC is what one would expect, but the only time I have seen it that high is when I have put 5V in via the serial connection (which is generally a bad idea).

kbaras, 2018-04-24:
Thank you, this was the cause, I am sure!
When you say " I try and avoid using GPIO for anything other than a momentary push button" do you think of GPIO0, or all the GPIO inputs?