arendst/Tasmota #2498: Tasmota mqtt and json problem

hugodelaroza, 2018-04-20:
Hi, I'm using a wemos d1 mini with tasmota and added some sensors. I publish the data over MQTT and read it from Home Assistant, mut in this case, I don't know why, the device doesn't publish the data. Here goes the console log:

21:36:08 MQT: cmnd/D1mini01PIR/POWER = OFF
21:37:13 RSL: tele/wemos01/STATE = {"Time":"2018-04-19T21:37:13","Uptime":"0T02:40:16","Vcc":2.980,"Wifi":{"AP":1,"SSId":"dlR","RSSI":38,"APMac":"6C:3B:6B:88:4F:32"}}
21:37:13 RSL: tele/wemos01/SENSOR = {"Time":"2018-04-19T21:37:13","Switch2":"OFF","DHT11":{"Temperature":27.0,"Humidity":59.0},"BH1750":{"Illuminance":9},"TempUnit":"C"}
21:39:23 MQT: cmnd/D1mini01PIR/POWER = ON
21:39:26 MQT: cmnd/D1mini01PIR/POWER = OFF
21:42:09 MQT: cmnd/D1mini01PIR/POWER = ON
21:42:12 MQT: cmnd/D1mini01PIR/POWER = OFF
21:42:13 RSL: tele/wemos01/STATE = {"Time":"2018-04-19T21:42:13","Uptime":"0T02:45:16","Vcc":2.980,"Wifi":{"AP":1,"SSId":"dlR","RSSI":36,"APMac":"6C:3B:6B:88:4F:32"}}
21:42:13 RSL: tele/wemos01/SENSOR = {"Time":"2018-04-19T21:42:13","Switch2":"OFF","DHT11":{"Temperature":27.0,"Humidity":58.0},"BH1750":{"Illuminance":9},"TempUnit":"C"}
21:43:40 MQT: cmnd/D1mini01PIR/POWER = ON
21:43:42 MQT: cmnd/D1mini01PIR/POWER = OFF

As you can see, there are some "RSL" lines, I don't know that that means, but this json message doesn't reach the HA.

Can someone helpme with this?

Frogmore42, 2018-04-20:
Your signal strength is somewhat low: "RSSI":38
But that probably isn't the main issue, since you seem to be getting the MQTT commands

Status 0 will indicate which components you are using since some people have had issues with some combinations.

curzon01, 2018-04-20:
Provide `Status 0` output.
It can be that you have played wth MQTTClient (set it different to default `"DVES_%06X"`) or you use lib 2.4.1 which causes the same problem described in other [issue](https://github.com/arendst/Sonoff-Tasmota/issues/2417#issuecomment-380429229) or... so pls provide `Status 0`

arendst, 2018-04-20:
RSL means Result. It only shows when no MQTT connection is active. Strange thing is you are still able to send MQTT Power On/Off for reasons unknown to me.

hugodelaroza, 2018-04-20:
Hi, this Is the data you asked for:
18:00:30 RSL: stat/wemos01/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"wemos01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:00:30 RSL: stat/wemos01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/wemos01.ino.bin","RestartReason":"External System","Uptime":"0T00:01:31","StartupUTC":"2018-04-20T20:58:59","Sleep":0,"BootCount":68,"SaveCount":1897,"SaveAddress":"3F6000"}}
18:00:30 RSL: stat/wemos01/STATUS2 = {"StatusFWR":{"Version":"5.12.0l","BuildDateTime":"2018-04-19T22:44:08","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
18:00:30 RSL: stat/wemos01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["dlR","RimayaInfomatix"],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
18:00:30 RSL: stat/wemos01/STATUS4 = {"StatusMEM":{"ProgramSize":553,"Free":2516,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2}}
18:00:30 RSL: stat/wemos01/STATUS5 = {"StatusNET":{"Hostname":"wemos01-4447","IPAddress":"192.168.0.118","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"EC:FA:BC:0E:B1:5F","Webserver":2,"WifiConfig":3}}
18:00:30 RSL: stat/wemos01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.68","MqttPort":1883,"MqttClientMask":"wemos01","MqttClient":"wemos01","MqttUser":"infomatix","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:00:30 RSL: stat/wemos01/STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 20 21:00:30 2018","Local":"Fri Apr 20 18:00:30 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"13:01","Sunset":"27:00"}}
18:00:30 RSL: stat/wemos01/STATUS10 = {"StatusSNS":{"Time":"2018-04-20T18:00:30","Switch2":"OFF","DHT11":{"Temperature":26.0,"Humidity":65.0},"BH1750":{"Illuminance":4},"TempUnit":"C"}}
18:00:30 RSL: stat/wemos01/STATUS11 = {"StatusSTS":{"Time":"2018-04-20T18:00:30","Uptime":"0T00:01:31","Vcc":2.982,"Wifi":{"AP":1,"SSId":"dlR","RSSI":96,"APMac":"78:8A:20:5D:A7:A9"}}}
18:02:00 MQT: tele/wemos01/UPTIME = {"Time

Thanks a lot

hugodelaroza, 2018-04-20:
I flashed tasmota 5.11 and woks great!

Thanks allí oficina you

closed 2018-04-20 by hugodelaroza