arendst/Tasmota #2500: Configuring nodemcu led-switch-relay

stephanfx, 2018-04-20:
[image]

I have the above circuit setup with the nodemcu. I have it where the leds lights up when the relay is switched on, the problem I am experiencing is that the Led1 in the drop down lights up no matter what relay is switched to on with the button. Is it possible to set that Led1 switches on when relay 1 is switched on, Led2 is switched on when relay2 is switched on.

Any advise would be appreciated.

Jason2866, 2018-04-20:
What i dont understand, why not connecting the LEDs on the relay GPIO?

stephanfx, 2018-04-20:
The relay board is this one [https://www.sunfounder.com/4-channel-5v-relay-shield-module.html](https://www.sunfounder.com/4-channel-5v-relay-shield-module.html). I can't see gpios, also, im a noob 😀

closed 2018-04-22 by stephanfx