arendst/Tasmota #2504: Presenting password exposure through RESULT on. Mqtt.

stefanbode, 2018-04-20:
I just saw a very simplified video how to hack Tasmoto on YouTube. Most of it bullshit, if you secure your mosquito server like explained in the Wiki. Anyhow I would vote for disabled password transmition on serial and Mqtt for the ssid password , mqtthost password, fingerprint, Mqttuser password.

If the user do not remember, his problem. Passwords should not transmitted into cleartext by whatever method. All modern software system do not net you access a password in cleartext once after entered.

Leseratte10, 2018-04-20:
If you have access to the device, you can get the passwords. If you have web access, just export the config and you have the password. If you have access, it's game over anyways.

stefanbode, 2018-04-21:
In this case i mentioned you only need access to the Mqtt broker. This is adifferentb thing. Also why everything should be lost if you can grab the device. Can you easily dump the flash to a disc and Analyse. As far as know not. For example if using TLS Mqtt and Webserver is off.  Hm, ok does not help really, because serial connection allows me to enable it again. Ok password must go.away from all communications. Also web. I missed the http password for the remove list.

Leseratte10, 2018-04-21:
That doesn't help. Serial connection, enable webserver, go to "Configuration" -> "Backup Configuration", and you have a file with all the settings, including all passwords. Took me about five minutes to find the part of the source code that is responsible for obfuscating and deobfuscating the backup file and write my own C program to decode it, then you have all passwords in plain text. 

Http password is also removed in my PR #2505.

Leseratte10, 2018-04-21:
The webserver password is useless at the moment. It restricts access to all the menus, but not to /dl, the page to download the current config with all passwords.

arendst, 2018-04-21:
Whow!  I'm glad I just removed command CfgDump...