arendst/Tasmota #2508: color <RGB> not accurate

gemu2015, 2018-04-21:
latest tasmota development  release

=> color 102080

08:05:18 CMD: color 102080
08:05:18 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":50,"Color":"0F1F7F","HSBColor":"231,88,50","Channel":[6,12,50]}

=> color 4473c3

08:06:40 CMD: color 4473c3
08:06:40 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":76,"Color":"4272C1","HSBColor":"217,66,76","Channel":[26,45,76]}

RESULT color is different , same result with channels

color should be stored bit accurate!

arendst, 2018-04-22:
As designed.

The color is always adjusted to it's nearest dimmer value between 0 and 100.

I do have to congratulate though if you see the color difference between 4473c3 and 4272c1

gemu2015, 2018-04-22:
OK, i expected the opposite behavior accurate RGB and rounded DImmer.value.