arendst/Tasmota #2509: Remote acccess

gabryk91, 2018-04-21:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x ] What hardware you are using
Sonoff basic

@arendst I'm trying to connect to my tasmota from outside, I did all the necessary configuration on my router from an x port to port 80 of the tasmota, but I can't see the web server, I had searched in the webserver.ino for the suggested lines, but I can't find those (I guess they were changed in recent updates). Can you provide me some direction to solve this problem? I understand the warning about opening the tasmota to the outside world, but still I want to do it with all the precautions, thanks

edit: if I would edit the webserver.ino as suggested by @splodgemcroo how can I send it to tasmota then?

maciejtarmas, 2018-04-21:
Have you tried accessing the Sonoff from a network outside of your router? A mobile connection, your friend’s place, whatever?

Most consumer routers won’t let you access your WAN IP if you’re on the LAN side. Google the term „NAT reflection”.

gabryk91, 2018-04-21:
I have other stuff reachable from the outside and i can reach them with public ip address even from my wifi. Anyway I did tried to connect from the 4g of my cell with no luck. Other thoughts?

gabryk91, 2018-04-23:
Yes

closed 2018-06-22 by stale[bot]