arendst/Tasmota #2512: external switches sonoff ch4 r2

xunil79, 2018-04-21:
opening the plastic shell I see the pins for the UART but no others pins for GPIO. I need to use GPIO for some switches. Is there any way to use external switches?

Leseratte10, 2018-04-21:
You could use RX and TX (GPIO 3 and GPIO 1) as switches in Tasmota, if you disable serial logging.

xunil79, 2018-04-21:
only 2 switches?

Frogmore42, 2018-04-22:
There are 4 switches on the device, you could extend the wires and put switches on them, as long as you don't need to use the ones on the device.

It looks like there are 3 user gpios on the pro.  If you need more, get another esp8266 based device and you can have more.

Leseratte10, 2018-04-22:
If you solder directly to the switches in the device, these only work for momentary push buttons, same as the ones on the device. If you need the more specific SwitchMode settings to work, you only have two (without soldering) or three (with soldering).

closed 2018-06-21 by stale[bot]