arendst/Tasmota #2516: Switch ON and Then OFF when restart

osmassoglia, 2018-04-21:
Make sure these boxes are checked before submitting your issue - Thank you!

I'm using the 1ch Relay 5V / 12V with tasmota sonoff in my garage door, the problem is when the sonoff restart for any reason send the signal Powe On and Later Power Off this is a problem for my because the gate open.

Is possible replace in the code this functionality to disable this function at retstart?

Best Regards

Frogmore42, 2018-04-21:
Tasmota does not do that by default.  You must have it configured or part of a system that is causing it to behave this way.  This is the reason there is the set of questions in the template issue.  You will need to provide that information for anyone to help you.  Also what home automation system are you using.  At least one of them is known to cause issues like this.

osmassoglia, 2018-04-22:
Here the console log

00:00:00 Project sonoff Sonoff (Topic sonoff-rele-1, Fallback sonoff-rele-5v, GroupTopic sonoffs) Version 5.12.0m-2_4_0
00:00:00 WIF: Connecting to AP1 babilonia in mode 11N as sonoff-rele-1-7443...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff-rele-1-7443.local with IP address 192.168.1.143
00:00:04 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff-rele-1/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff-rele-1/POWER = 
00:00:05 MQT: tele/sonoff-rele-1/INFO1 = {"Module":"Sonoff T1 1CH","Version":"5.12.0m","FallbackTopic":"sonoff-rele-5v","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff-rele-1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-rele-1-7443","IPAddress":"192.168.1.143"}
00:00:05 MQT: tele/sonoff-rele-1/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/sonoff-rele-1/RESULT = {"POWER":"OFF"}
00:00:06 MQT: stat/sonoff-rele-1/POWER = OFF
00:00:06 MQT: stat/sonoff-rele-1/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff-rele-1/POWER = ON
20:56:16 MQT: stat/sonoff-rele-1/RESULT = {"POWER":"OFF"}
20:56:16 MQT: stat/sonoff-rele-1/POWER = OFF
20:56:22 MQT: tele/sonoff-rele-1/STATE = {"Time":"2018-04-21T20:56:22","Uptime":"0T00:00:13","Vcc":3.409,"POWER":"OFF","Wifi":{"AP":1,"SSId":"babilonia","RSSI":36,"APMac":"C4:E9:84:E9:23:82"}}

You are right, the problem is caused by MQTT y try removing configuration and the device work perfect...  i need configure the device with MQTT, anyone know how can disable this functionality?

Frogmore42, 2018-04-22:
Status 0?
There are lots of ways things can be configured.  The relay can be configured to remember, or be on or off on reset.  Which way do you have it configured?

MQTT can be configured to retain the last state, which can conflict with the configuration above.

Once again, what system for Home Automation are you using?
Why do you need MQTT?

osmassoglia, 2018-04-22:
I'm using OpenHab

Frogmore42, 2018-04-22:
I don't know OpenHab, but either you have MQTT retain set (maybe OpenHab needs this) or OpenHab is not dealing well with the reconnect.

I use MQTT, but I am using NodeRED for control.  I never have a problem when they reconnect (and they also do it very rarely).

I have seen lots of reports of problems with people using various HA solutions.  I believe some make it very hard to use Tasmota.

Jason2866, 2018-04-22:
I use OpenHab too. OpenHab if configured correct, and this is the default setup out of the box.
OpenHab does not store and send messages (retain).
So your setup of the sonoff is buggy or you changed the default setup in OpenHab.
Reset your sonoff device and begin from a fresh setup.
Do everything @arendst described in the wiki and you will get a working solution.
I have >20 Tasmota devices and NONE have a strange behaviour after power loss, switching off WLAN
or reconnects. I have devices with uptime without any reconnect > 60 days.

Jason2866, 2018-04-22:
Do you use the included mosquitto broker? Which device do you use for OpenHab?
If a clean setup for the sonoff devices doesnt help go to:
community.openhab.org

avegao, 2018-05-20:
I have the same problem with a device without any automation system (just flashed and WiFi configuration). When I restart the device, the relay toggle.

Also when I change the status of the relay, the device stopped responding and a led blinks indefinitely until I disconnect it.

Here is my log:

```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_838803, GroupTopic sonoffs) Version 5.14.0-2_4_1
00:00:00 WIF: Connecting to AP1 ASUS-WIFI in mode 11N as sonoff-2051...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-2051.local with IP address 192.168.1.158
00:00:05 RSL: INFO1 = {"Module":"1 Channel","Version":"5.14.0","FallbackTopic":"DVES_838803","GroupTopic":"sonoffs"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-2051","IPAddress":"192.168.1.158"}
00:00:05 RSL: INFO3 = {"RestartReason":"Software/System restart"}
22:07:05 RSL: RESULT = {"POWER":"OFF"}
22:07:05 RSL: POWER = OFF
22:07:12 RSL: STATE = {"Time":"2018-05-20T22:07:12","Uptime":"0T00:00:13","Vcc":3.462,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ASUS-WIFI","RSSI":100,"APMac":"F0:9F:C2:CC:F7:53"}}
22:07:20 RSL: RESULT = {"POWER":"ON"}
22:07:20 RSL: POWER = ON
22:07:21 RSL: RESULT = {"POWER":"OFF"}
22:07:21 RSL: POWER = OFF
```

Frogmore42, 2018-05-20:
Status 0?

The log indicated you are not connected to MQTT.  Not having an MQTT broker requires non default configuration.

closed 2018-07-10 by ascillato2

d0v1k, 2019-03-05:
Did you manage to fix it? i'm experiencing the same issue.

jrailton, 2020-01-26:
I'm having the same issue as well, Sonoff SV (inching/self locking) with Tasmota 8.1.0 - every time device restarts, the relay toggles on and off again.

Mine IS connected to an MQTT broker but there is no retained switch state (even if i shut the broker down and restart the device, the relay will still toggle)

mpuff, 2020-05-10:
i have the same problem, i think since tasmota v7 and 8, bevor, i never had problems with this, i noticed it when I made firmware updates, in version 6 the sonoffs did not set the power state to off and then on, so it stayed on even during a reboot

Jason2866, 2020-05-10:
Guys please use latest development version. 
There are changes made that  NO GPIO blinking is there anymore.
Or wait for Tasmota 8.3 release will be on 14th of May

jrailton, 2020-05-11:
> Guys please use latest development version.
> There are changes made that NO GPIO blinking is there anymore.
> Or wait for Tasmota 8.3 release will be on 14th of May

Appreciated, thanks. I noticed mine was related to switch losing wifi signal, starting AP mode and rebooting every 3 minutes until wifi signal was recovered. Changed mine to WifiConfig 4 and will see if that helps :)

Heisenberg2980, 2021-11-09:
> Appreciated, thanks. I noticed mine was related to switch losing wifi signal, starting AP mode and rebooting every 3 minutes until wifi signal was recovered. Changed mine to WifiConfig 4 and will [...]

I am having the same issue, devices lose connection every day at the same time (I think it is my wifi extender rebooting every 24h) and they start AP mode and reboot after 3 minutes (which is correct as I setup WifiConfig 2). I am happy with the devices switching to AP mode and rebooting when they lose connection to wifi, but I would like to avoid the blink, is there any way to avoid the devices switching off and on in this case?

I know I could change to WifiConfig 4 but I like the idea that if the wifi is gone I still can connect to the devices and interact with them.

Heisenberg2980, 2021-11-10:
Just in case anyone has the same issue of the light "blinking" during the reboot, this is expected for chips ESP8266/ESP8285. More details in https://github.com/arendst/Tasmota/issues/13634