arendst/Tasmota #2517: xsns_05_ds18x20.ino for 5 DS bug for DOMOTICZ

roblad, 2018-04-22:
Hi,

I set 5 DS on Sonoff CH4 Pro and I wanted to use domoticz, and only one DS is reported by domoticz mqtt:

domoticz/in {"idx":345,"nvalue":0,"svalue":"27.69","Battery":200,"RSSI":10}

I tried set next IDX but nothing is reported for other 4 DS.

I guess that something is wrong in here:

#ifdef USE_DOMOTICZ
        if ((0 == tele_period) && domoticz_flag) {
          DomoticzSensor(DZ_TEMP, temperature);
          domoticz_flag = false;
        }
#endif  // USE_DOMOTICZ

Result:

tele/podlewaczka/SENSOR {"Time":"2018-04-22T07:55:10","DS18B20-1":{"Id":"0000054BDDF6","Temperature":27.69},"DS18B20-2":{"Id":"0000054BDDF6","Temperature":28.81},"DS18B20-3":{"Id":"0000054C3854","Temperature":26.50},"DS18B20-4":{"Id":"0000054CCEC4","Temperature":43.69},"DS18B20-5":{"Id":"0000054D6492","Temperature":21.44},"AM2301":{"Temperature":22.70,"Humidity":37.2},"Gas":{"Type":"CLEANAIR","ppm":0},"TempUnit":"C"}

domoticz/in

domoticz/in {"idx":345,"nvalue":0,"svalue":"27.69","Battery":200,"RSSI":10}
domoticz/in {"idx":346,"nvalue":0,"svalue":"22.70;37.2;2","Battery":200,"RSSI":10}
domoticz/in {"idx":351,"nvalue":0,"svalue":"CLEANAIR","Battery":200,"RSSI":10}
domoticz/in {"idx":353,"nvalue":0,"Battery":200,"RSSI":10}

roblad, 2018-04-22:
Hi,

it means that only one DS will be reported to domoticz even when I have 5 ?

localhost61, 2018-04-22:
@roblad you may play with pullup resistor value to enhance detection. For one DS18B20 it should be 4.7kΩ. But for calibration purpose, I recently made an experiment with an Arduino ProMini (under 5V, not 3.3V) and as much as 21 DS18B20, with different cable lengths and got better detection results with a resultant value of 1.5kΩ.

roblad, 2018-04-22:
I will try

roblad, 2018-04-22:
Did not help

arendst, 2018-04-29:
Use rules to as explained in #2541

closed 2018-04-29 by arendst

roblad, 2018-04-30:
Hi,

Why it does not work ?

19:38:14 RSL: Grupa 0, Indeks 1, Komenda RULE, Data on tele-DS18B20-1#temperature do publish domoticz/in {"idx":120,"nvalue":0,"svalue":"%value%"} endon
19:38:14 MQT: stat/podlewaczka/RESULT = {"Rule":"ON","Once":"OFF","Rules":"on tele-DS18B20-1#temperature do publish domoticz/in {"idx":120,"nvalue":0,"svalue":"%value%"} endon"}