arendst/Tasmota #2518: tls in current version fails

or78rXxZ, 2018-04-22:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

I compiled 5.12.0 wth tls and it gets an error but works.
Apr 22 05:36:08 sonoff_b6-7436 ESP-MQT: Verify TLS fingerprint...
Apr 22 05:36:08 sonoff_b6-7436 ESP-MQT: Insecure connection due to invalid Fingerprint

I compiled  the current version 5.12.0m with tls and it fails
Apr 22 05:50:19 sonoff_b6-7436 ESP-MQT: Attempting connection...
Apr 22 05:50:19 sonoff_b6-7436 ESP-MQT: Verify TLS fingerprint...
Apr 22 05:50:19 sonoff_b6-7436 ESP-MQT: Failed

arendst, 2018-04-22:
As you might have noticed in 5.12.0 it continues the MQTT connection even while the fingerprint is wrong.

In version 5.12.0m a wrong fingerprint just aborts MQTT connection as a security measure.

You will have to get a valid fingerprint to connect to your MQTT server.

or78rXxZ, 2018-04-22:
I see the problem but not a clue where the problem lies

06:07:19 MQT: Attempting connection...
06:07:19 MQT: Verify TLS fingerprint...
[hostByName] Host: 192.168.2.16 is a IP!
:ref 1
Alert: close notify
:ref 2
State:	sending Client Hello (1)
:wr 93 93 0
:wrc 93 93 0
:sent 93
:rn 1460
:rd 5, 1460, 0
:rdi 1460, 5
:rd 74, 1460, 5
:rdi 1455, 74
State:	receiving Server Hello (2)
:rd 5, 1460, 79
:rdi 1381, 5
:rd 1376, 1460, 84
:r:rch 1460, 344
di 1376, 1376
:c 1376, 1460, 1804
:rd 335, 344, 0
:rdi 344, 335
State:	receiving Certificate (11)
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		g2
Organization (O):		Internet Widgits Pty Ltd
Country (C):			AU
State (ST):			Some-State
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		g3
Organization (O):		Internet Widgits Pty Ltd
Country (C):			AU
State (ST):			Some-State
Not Before:			Fri Apr 20 12:07:25 2018
Not After:			Mon Apr 17 12:07:25 2028
RSA bitsize:			2048
Sig Type:			SHA256
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		g3
Organization (O):		Internet Widgits Pty Ltd
Country (C):			AU
State (ST):			Some-State
Basic Constraints:		critical, CA:TRUE, pathlen:10000
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		g3
Organization (O):		Internet Widgits Pty Ltd
Country (C):			AU
State (ST):			Some-State
Not Before:			Fri Apr 20 12:03:53 2018
Not After:			Mon Apr 17 12:03:53 2028
RSA bitsize:			2048
Sig Type:			SHA256
:rd 5, 344, 335
:rdi 9, 5
:rd 4, 344, 340
:rdi 4, 4
:c0 4, 344
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
:wr 267 267 0
:wrc 256 267 0
:wrc 11 11 0
:wr 6 6 0
:wrc 6 6 0
State:	sending Finished (16)
:wr 85 85 0
:wrc 85 85 0
:sent 267
:sent 91
:rn 91
:rd 5, 91, 0
:rdi 91, 5
:rd 1, 91, 5
:rdi 86, 1
:rd 5, 91, 6
:rdi 85, 5
:rd 80, 91, 11
:rdi 80, 80
:c0 80, 91
State:	receiving Finished (16)
cert FP: 49 98 BE 2D 34 4E B7 B5 0C 31 98 45 39 13 43 CD D6 3F 56 DA 
test FP: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
fingerprint doesn't match
cert FP: 49 98 BE 2D 34 4E B7 B5 0C 31 98 45 39 13 43 CD D6 3F 56 DA 
test FP: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
fingerprint doesn't match
06:07:20 MQT: Failed
:ur 2
:close

or78rXxZ, 2018-04-22:
I followed your instructions "Secure Setup"
On my mqtt server
# openssl s_client -connect localhost:8883 < /dev/null 2>/dev/null | openssl x509 -fingerprint -noout -in /dev/stdin
SHA1 Fingerprint=49:98:BE:2D:34:4E:B7:B5:0C:31:98:45:39:13:43:CD:D6:3F:56:DA

arendst, 2018-04-22:
I did not write the Secure Setup part but....

As your status 0 output is incomplete I cannot see what MQTT server you are trying to reach but using a localhost as MQTT server and TLS seems overkill to me. Anyway the fingerprint as to be known to both your device AND your MQTT server so the part regarding Mosquitto SSL configuration seems relevant to me.

Leseratte10, 2018-04-22:
This: 

    State: receiving Finished (16)
    cert FP: 49 98 BE 2D 34 4E B7 B5 0C 31 98 45 39 13 43 CD D6 3F 56 DA
    test FP: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    fingerprint doesn't match
seems to mean that you didn't add the fingerprint to the user_config.h before compiling.

arendst, 2018-04-22:
There is no need to add anything to the user_config.h

When do people finally grasp the use of commands and persistent settings....

or78rXxZ, 2018-04-22:
You are correct I added it to user_config_override.h
Do I need to move all of my configuration back to user_config.h

closed 2018-06-21 by stale[bot]

fr34kyn01535, 2018-08-04:
If others stumble upon the issue, make sure to also have the certificate cn match the mqtt hostname. Took a while until i figured, reading how it connects via WiFiClientSecure::verify.