arendst/Tasmota #2519: Sonoff Pow JSON Enery format in MQTT message causes problem with JSONPath

beamish04, 2018-04-22:
I am configuring Sonoff POW modules in Openhab2 using MQTT / Mosquitto. The powe is running Tasmota firmware  5.12.0 

I am having troubles to convert the value for power consuption, i.e. Yesterday or Today. 
I am not sure if this issue is a Tasmota related issue or more related to the transformation module 
JSONPath  in Openhab. 

My observation is that JSONPath is cutting off trailing zeros of a value when it is interpreted as a number. 
As the structure of the MQTT message is sending the value for Yesterday as number and not as string in quotes I see this problem as follows: 

This is the message which Sonoff POW is sending per console and received by the MQTT broker: 

`12:21:08 MQT: tele/pow1/SENSOR = {"Time":"2018.04.22 12:21:08","ENERGY":{"Total":29.214,"Yesterday":1.200,"Today":0.10,"Period":0,"Power":0,"Factor":0.00,"Voltage":219,"Current":0.000}}`

With the obove message the values for Yesterday is interpreted as 1.2 instead of 1200

This seems not only to be true for the Openhab JSONPath implementation. I checked this as well with 
[http://www.jsonquerytool.com/#/JSONPath](url)

[image]

Assuming the POW has logged power consumption of 1207 W the day before then the parsing with JSONPath works fine. But when the power consumption was only 1200 W then the value gets truncated to 12 by JSONPath. 

Would it make sense to send such values as strings instead of numbers? This would differenticate 
int numbers as such for values like Total, Yesterday, Today as INT from values as float like temperature. The JSON definition for Temperature is also using the number representation but the same is used for the presentation of thousands for other parameters

```
**Sonoff Module: BMP280**
{
  "StatusSNS": {
    "Time": "2018-02-10T22:46:34",
    "BMP280": {
      "Temperature": 80.9,
      "Pressure": 984.4
    }
  }
}
```
**Sonoff Module: POW**
"StatusPTH": {
    "PowerLow": 0,
    "PowerHigh": 0,
    "VoltageLow": 0,
    "VoltageHigh": 0,
    "CurrentLow": 0,
    "CurrentHigh": 0
  },
  "StatusSNS": {
    "Time": "2018.02.04 23:17:01",
    "ENERGY": {
      "Total": 3.185,
      "Yesterday": 3.058,
      "Today": 0.127,
      "Power": 0,
      "Factor": 0.00,
      "Voltage": 221,
      "Current": 0.000
    }
  },

Or is there a possiblility to change the representation of tousands to not use the dot? 
If not sending the value as string would make it possible to run customized transformation 
of the string, i.e. remove the dot and parse the value to Int of Float, whatever is needed. Right now I did not find a solution to retrieve the complete value from the MQTT message as the number gets manupilated by JSONPath upfront. 

many thanks & Regards,

arendst, 2018-04-22:
The dot is the decimal separator NOT the thousand indicator. It returns kW not W

beamish04, 2018-04-22:
Thanks Theo for your immediate response and the clarification. You made my day. I was looking so much in the wrong area that I overlooked the obvious. 

Thanks a lot!!!

closed 2018-04-22 by beamish04