arendst/Tasmota #2520: [upload] Error -1

nicolocandiani, 2018-04-22:
I'm trying to install Tasmota on my Sonoff basic using Atom and PlatformIO.
I installed the drivers form the serial port adapter and when I try to upload everything to the sonoff I get: `[upload] Error -1`

The console says:
`Auto-detected: COM3`
`Uploading .pioenvs\sonoff\firmware.bin`
`warning: espcomm_sync failed`
`error: espcomm_open failed`
`error: espcomm_upload_mem failed`
`*** [upload] Error -1`

Leseratte10, 2018-04-22:
That just means it can't find the sonoff. Is it in flash mode? (did you hold down the button while plugging it in?), did you try swapping TX and RX?

ascillato, 2018-06-02:
Hi @nicolocandiani 

Could you manage to flash? If yes, please close this issue. Thanks.

closed 2018-06-02 by nicolocandiani