arendst/Tasmota #2521: IPv6 Support?

mwuertinger, 2018-04-22:
The [Internet Protocol version 6](https://en.wikipedia.org/wiki/IPv6) was designed 20 years ago. Unfortunately many devices (including virtually all IoT devices) still do not support it. IPv4 addresses are getting [extremely scarce](https://en.wikipedia.org/wiki/IPv4_address_exhaustion) these days and many ISPs no longer allocate public IPv4 addresses for their customers ([Carrier grade NAT](https://en.wikipedia.org/wiki/Carrier-grade_NAT)) which is already causing connectivity issues for many people including myself. Therefore we should all strive for 100% IPv6 support.

Would it be possible to implement IPv6 on Sonoff devices or is there anything in particular that stands in the way?

Leseratte10, 2018-04-22:
Last I checked, the upstream libraries didn't support IPv6 yet: esp8266/Arduino#638 
I found this: espessif/arduino-esp32#1261, but I don't know if this applies to the ESP8266. I might check that soon, as I am interested in IPv6 support as well.

d-a-v, 2018-09-14:
Check again https://github.com/esp8266/Arduino/issues/638

akram, 2020-01-15:
Ardui core now supports IPv6 : https://github.com/esp8266/Arduino/pull/5136

can we re-open thins one ?

ascillato, 2020-01-15:
> can we re-open thins one ? 

No because it is already implemented in Tasmota. You have to compile by yourself enabling IPV6 support in platformio.ini

If you need further assistant, please, do not hesitate on asking on the Support Chat. Thanks.

akram, 2020-01-17:
Hi @ascillato , thank you for your quick answer.
I had a look at platformio.ini and I didn't find a self exaplanatory commented out parameter to enable it.
That would be good if you could tell which one it is?

akram, 2020-01-17:
my bad , it looks like related to this:
```
; lwIP 1.4
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
; lwIP 2 - Low Memory
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
; lwIP 2 - Higher Bandwidth
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
; lwIP 2 - Higher Bandwidth Low Memory no Features
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH
; lwIP 2 - Higher Bandwidth no Features (Tasmota default)
                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH
; lwIP 2 - Higher Bandwidth IPv6 (use ONLY if you need IPv6, experimental!)
;                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_IPV6_HIGHER_BANDWIDTH

```

ascillato, 2020-01-17:
yes, exactly. 

If you need help on using those, you can use the support chat if you want.

ascillato, 2020-01-17:
Anyway, as the comment explains, it is experimental because the support is on the Arduino Core. You should use the stage core that have several fixes for that, but still experimental.

Use it, if you really need ipv6 in your local network. ipv4 for a local network is reliable and works for all usecases.

freifunkerffm, 2020-09-23:
Hello,
from the post above I understand that TASMOTA is supporting IPV6 in general.
I have already a working platformio-core setuo where I can compile TASMOTA with different features with success!
But I fail with IPV6 :-/
In platformio.ini I have set:
; lwIP 2 - Higher Bandwidth IPv6 (use ONLY if you need IPv6, experimental!)
-DPIO_FRAMEWORK_ARDUINO_LWIP2_IPV6_HIGHER_BANDWIDTH

But it fails to work on a NODEMCU V3 board.
The serial-log from the device shows:
00:00:00 CFG: aus Flash geladen am F4, zählen 8
00:00:00 QPC: Count 1
00:00:00 Projekt tasmota Tasmota Version 8.5.0.1(tasmota)-2_7_4_1
00:00:00 HTP: Web-Server aktiv bei tasmota_40FABB-6843 mit IP-Adresse (IP unset)
00:00:01 RSL: tele/tasmota_40FABB/INFO1 = {"Module":"Sonoff Basic","Version":"8.5.0.1(tasmota)","FallbackTopic":"cmnd/DVES_40FABB_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:01 RSL: tele/tasmota_40FABB/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_40FABB-6843","IPAddress":"(IP unset)","IPv6Address":""}
00:00:01 RSL: tele/tasmota_40FABB/INFO3 = {"RestartReason":"Software/System restart"}
00:00:01 RSL: stat/tasmota_40FABB/RESULT = {"POWER":"OFF"}
00:00:01 RSL: stat/tasmota_40FABB/POWER = OFF
00:00:01 RSL: homeassistant/discovery/807D3A40FABB/config = {"ip":"(IP unset)","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota_40FABB-6843","mac":"807D3A40FABB","md":"Sonoff Basic","ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"8.5.0.1","t":"tasmota_40FABB","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"btn":[1,0,0,0],"so":{"11":0,"13":0,"17":0,"20":0,"30":0,"37":1,"68":0,"73":0,"80":0},"lt_st":0,"ver":1}
00:00:05 RSL: tele/tasmota_40FABB/STATE = {"Time":"1970-01-01T00:00:05","Uptime":"0T00:00:07","UptimeSec":7,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":8,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"lalala.net","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100,"Signal":31,"LinkCount":1,"Downtime":"0T00:00:01"}}
00:00:06 QPC: Reset

So somehow IPV6 is mentioned but for it looks like I miss some buld-parameters.
The documentation on having TASMOTA IP6 is quite weak.

Has anyone compiled TASMOTA with IPV6 and can give dedicated hints what settings in platformio.iniare necessary?
I am also curious if ESP-8266 is then turned into a single IPV6 stack or into a dual stack (IP4/IP6) device?

Kai

altelch, 2020-09-23:
Hi,

no, you didn't miss a build-parameter. Just did a build on my own 
including #pragma message to see if #ifdef LWIP_IPV6 is set. And yes, it 
is but the startup seems to be messed up. Normaly IPv6 was DualStack and 
you should get at least an IPv4 address but now there is no address at 
all. I have to admit that this code was implemented in october 2019 and 
since then I didn't check this part. There where a lot of changes 
(including ESP32 support) since then. First step is to check 
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples/IPv6/IPv6.ino 
what is the example code of the arduino framework. If this works it 
needs some digging into the current tasmota code...can't promise to find 
some time, soon.

Greetings,

                   Altelch.

LWIP_IPV6

d-a-v, 2020-09-23:
That example will work ;-)

altelch, 2020-09-23:
Ok,

I'll trust you ;-) Then I need to find what's got messed up and when...

Thx,

                 Altelch.

altelch, 2020-09-24:
Hi,

IPv6 works with 8.5.0. Somehow platformio.ini isn't evaluated after changing DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH to DPIO_FRAMEWORK_ARDUINO_LWIP2_IPV6_HIGHER_BANDWIDTH

Did also put a
#pragma message ("IPv6 compile time enabled!")
into support_wifi.ino in WifiBegin after #ifdef LWIP_IPv6 to see if this branch is compiled. After a few trials gitpod started to compile the whole project again and this time I got my above message during compile. The firmare works as expected:
00:00:05 WIF: Got IPv6 global address xxxx:xxxx:150:5ecf:7fff:fe88:xxxx

Greetings,

           Heiko.

freifunkerffm, 2020-09-24:
Hello Heiko,
I tested it again with a freschly checked out TASMOTA  :  git clone https://github.com/arendst/Tasmota.git 
I changed platformio.ini like that:
                            ;-DPIO_FRAMEWORK_ARDUINO_LWIP3_HIGHER_BANDWIDTH_LOW_FLASH
                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_IPV6_HIGHER_BANDWIDTH
and used no user_config_override.h
But compiled it with platformio and not with gitpod.
I also added your pragma message in the support_wifi.ino.
That is visible in the build log:
Retrieved `.pio/build/tasmota/libeec/ESP8266SdFat/FatLib/FatFileSFN.cpp.o' from cache
/home/k/platformio-coreIP6/TasmotaIPV6/tasmota/support_wifi.ino: In function 'void WifiBegin(uint8_t, uint8_t)':
/home/k/platformio-coreIP6/TasmotaIPV6/tasmota/support_wifi.ino:214:52: note: #pragma message: IPv6 compile time enabled!
       #pragma message ("IPv6 compile time enabled!")
                                                    ^
Retrieved `.pio/build/tasmota/libeec/ESP8266SdFat/FatLib/FatVolume.cpp.o' from cache

I pushed the bin via tasmotizer to the NodeMCU V3 board and got:
00:00:00 CFG: aus Flash geladen am F9, zählen 3
00:00:00 QPC: Count 1
00:00:00 Projekt tasmota Tasmota Version 8.5.0.1(tasmota)-2_7_4_1
00:00:00 HTP: Web-Server aktiv bei tasmota_40FABB-6843 mit IP-Adresse (IP unset)
00:00:01 RSL: tele/tasmota_40FABB/INFO1 = {"Module":"Sonoff Basic","Version":"8.5.0.1(tasmota)","FallbackTopic":"cmnd/DVES_40FABB_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:01 RSL: tele/tasmota_40FABB/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_40FABB-6843","IPAddress":"(IP unset)","IPv6Address":""}
00:00:01 RSL: tele/tasmota_40FABB/INFO3 = {"RestartReason":"External System"}
00:00:01 RSL: stat/tasmota_40FABB/RESULT = {"POWER":"OFF"}
00:00:01 RSL: stat/tasmota_40FABB/POWER = OFF
00:00:01 RSL: homeassistant/discovery/807D3A40FABB/config = {"ip":"(IP unset)","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota_40FABB-6843","mac":"807D3A40FABB","md":"Sonoff Basic","ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"8.5.0.1","t":"tasmota_40FABB","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"btn":[1,0,0,0],"so":{"11":0,"13":0,"17":0,"20":0,"30":0,"37":1,"68":0,"73":0,"80":0},"lt_st":0,"ver":1}
00:00:05 RSL: tele/tasmota_40FABB/STATE = {"Time":"1970-01-01T00:00:05","Uptime":"0T00:00:07","UptimeSec":7,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":8,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100,"Signal":31,"LinkCount":1,"Downtime":"0T00:00:01"}}
00:00:06 QPC: Reset

I noticed a new WLAN named ESP-40FABB I could join and found at 192.168.4.1 a new tasmota device to configure.
(no hint for that in the logging)
I just changed the WIFI credentials via GUI and got:

00:00:00 CFG: aus Flash geladen am F6, zählen 6
00:00:00 QPC: Count 1
00:00:00 Projekt tasmota Tasmota Version 8.5.0.1(tasmota)-2_7_4_1
00:00:00 HTP: Web-Server aktiv bei tasmota_40FABB-6843 mit IP-Adresse (IP unset)
00:00:01 RSL: tele/tasmota_40FABB/INFO1 = {"Module":"Generic","Version":"8.5.0.1(tasmota)","FallbackTopic":"cmnd/DVES_40FABB_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:01 RSL: tele/tasmota_40FABB/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_40FABB-6843","IPAddress":"(IP unset)","IPv6Address":""}
00:00:01 RSL: tele/tasmota_40FABB/INFO3 = {"RestartReason":"Software/System restart"}
00:00:01 RSL: homeassistant/discovery/807D3A40FABB/config = {"ip":"(IP unset)","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota_40FABB-6843","mac":"807D3A40FABB","md":"Generic","ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"8.5.0.1","t":"tasmota_40FABB","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"btn":[0,0,0,0],"so":{"11":0,"13":0,"17":0,"20":0,"30":0,"37":1,"68":0,"73":0,"80":0},"lt_st":0,"ver":1}
00:00:05 RSL: tele/tasmota_40FABB/STATE = {"Time":"1970-01-01T00:00:05","Uptime":"0T00:00:07","UptimeSec":7,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"lalala.net","BSSId":"FF:FF:FF:FF:FF:FF","Channel":1,"RSSI":100,"Signal":31,"LinkCount":1,"Downtime":"0T00:00:01"}}
00:00:06 QPC: Reset

After this the ESP-40FABB WIFI is not gone, but the ESP8266 detects the W-Lan Network.
So for me IPv6 is still not really usable
Is there a kind of command list to execute to get some more information about that.
I am happy to help, if I can
 
Kai

freifunkerffm, 2020-09-29:
Hi all,
I gave it another go, this time I compliled the tasmota Development Branch: https://gitpod.io#https://github.com/arendst/Tasmota/tree/development via gitpod.
Just changed one line in  platformio.ini , I replaced:
DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH 
with
DPIO_FRAMEWORK_ARDUINO_LWIP2_IPV6_HIGHER_BANDWIDTH

And compiled a pure tasmota with: "platformio run -e tasmota"
Used tasmotizer to flash, and got again something I have already seen:

CFG: Loaded from flash at F8, Count 4
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 8.5.0.1(tasmota)-2_7_4_1
00:00:00 HTP: Web server active on tasmota_40FABB-6843 with IP address (IP unset)
00:00:01 RSL: tele/tasmota_40FABB/INFO1 = {"Module":"Sonoff Basic","Version":"8.5.0.1(tasmota)","FallbackTopic":"cmnd/DVES_40FABB_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:01 RSL: tele/tasmota_40FABB/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_40FABB-6843","IPAddress":"(IP unset)","IPv6Address":""}
00:00:01 RSL: tele/tasmota_40FABB/INFO3 = {"RestartReason":"External System"}

So sorry I am not able to confirm that IPV6 is working on a TASMOTA device.

Kai

XenGi, 2022-02-09:
2 years went by and my tasmota devices still has no IPv6 support by default. It is still needed to compile tasmota by hand to enable that feature?

Raja4Shekar, 2022-02-23:
There are so many places where public ipv4 address is not provided by ISP due to the use of CGNAT, so in those circumstances, ipv6 is too much useful. So many people will get benefited if Tasmota Support ipv6. 

Not sure if this link helps anyway
https://www.keithh.com/network/ipv6-setup-esp8266/

redhat421, 2022-02-23:
I run Tasmota on a S31 w/ IPv6 enabled. It works.

I give it an address via SLAAC.
I have noted that if the router isn't handing out Router Advertisements, the device will hang for ~2 mins then come up with an IPv4 address only, so look out for that.

Thanks!

XenGi, 2022-02-23:
Honestly you don't need to fish for NAT or IPv4 shortage reasons to support IPv6. It's the current version of the internet protocol and it is at this point ober 20 years old. By not enabling it by default your basically building legacy systems.

So i think this ticket should be reopened. IPv6 support is still not enabled by default. I would rather use something that lacks IPv4 support than something without IPv6.

RedBlackThunder, 2022-02-23:
I would like an IPv6-only option even more than dual stack.

s-hadinger, 2022-02-23:
There is no plan for IPv6-only option in Tasmota. Why would you even want to not support IPv4 ?

XenGi, 2022-02-23:
> There is no plan for IPv6-only option in Tasmota. Why would you even want to not support IPv4 ?

Let me rephrase:
There is no plan for car only roads. Why would you even want to not support coaches.

kroon040, 2022-07-16:
Will there be ipv6 full support? Dual stack, with dhcpv6 and slaac? It's availiable in the esp-idf, so hopefullt soon in tasmota

sfromis, 2022-07-16:
As Tasmota is designed for internal networks, not the big bad Internet, the "real benefit" of IPv6 would be marginal at best. I certainly have no scarcity of IP addresses within the reserved ranges.

XenGi, 2022-07-18:
IPv6 has nothing to do with internal networks. If tasmota doesn't support the current version of the internet protocol it is basically legacy software. I really don't get this hostility towards IPv6. There is really no reason to not support it.
It's the standard way of communicating between IP capable devices and that since over 25 years!
Get your shit together. You're embarrassing yourself.

sfromis, 2022-07-18:
Not at all "hostility", just explaining why IPv6 is not seen as much of a priority. While I like the concept of getting everything on IPv6, that's not going to happen in the foreseeable future anyway, and has most value when it comes to using the scarce public IP addresses.

Anyway, Tasmota is not managing the IP support, that's happening in the frameworks outside of Tasmota code. It does not help to express your desires here.

barbudor, 2022-07-18:
@XenGi please respects the community guidelines.

Now on a pure technical aspect ESP8266 is *legacy* hardware with limited capabilities and the Espressif SDK is frozen AFAIK (no more updates /maintenance). Investing in IPv6 on ESP8266 sounds like a loss of time (not sure any vendor has ever supported it, may be shelly?) 

2 years ago, investing in ESP32 sounded like a loss of time. Even supporting Ethernet didn't looked useful. Now both are here. 
With ESP32 and more variants coming with more memory and flash, why not?
It's up to someone taking the task and proposing one (or probably more PR).
So if you are such an IPv6 enthusiast, just embrace that tasks and contribute.

sfromis, 2022-07-18:
No one said that "no one needs" or "not commonly used". IPv6 is very much needed and used, but all depends on use case.

s-hadinger, 2022-07-18:
There is no need to be aggressive when I simply ask why you need it.

The main use-case for Tasmota is to run on a local network and connect to a MQTT gateway on the same local network, which happen to be 99% IPv4 anyways. Some people (like me) connect to a remote MQTT gateway, but dual-stack does the job.

We have nothing against IPv6, by far. But on a local network it just adds unnecessary complexity when it comes to configuration, which is the reason we think most users won't use IPv6 anyways.

If you want IPv6 support, we are more than happy to get a Pull Request from you. Arduino and esp-idf seem to support IPv6 already. Be our guest, and have a nice day.

XenGi, 2022-07-28:
I'm not good at embedded programming. So as much as I would like to add IPv6 support myself, I don't see myself being able to do it. I guess I will look for an alternative solution for my network. I'm not willing to run dual stack only for two little iot devices.

Thx everyone for your input, sorry I went a little harsh. This topic is very important to me, so sometimes I go a bit far.
I hope tasmota will support current internet standards at some point.

bzfbd, 2022-11-27:
> We have nothing against IPv6, by far. But on a local network it just adds unnecessary complexity when it comes to configuration, which is the reason we think most users won't use IPv6 anyways.

I think that is plain old thinking.  I haven't had IPv4 on the LAN anymore for more than a decade.

It's a constant struggle to make people understand that IPv6 is not an option anymore but a must or otherwise things won't work for people.  And there are places in this world that are all IPv4 (private addresses) still and like those there are places where IPv4 is not provided anymore [unless you pay a lot of money] or places which implemented IPv6 and removed IPv4 to remove the complexity of dealing with two protocols at no benefit.
You can span up your IPv4 (private address) island but you just struggle with a lot of problems then as well.

I was trying to look into Tasmota but given these discussions I feel like I'll wait before investing.  I've fixed other esp32-idf based projects before but if there's further frameworks on top it's just a lot of hassle if they haven't been fixed before.  But then there are also people saying it works.  This is confusing.  Like @RedBlackThunder  IPv6-only options would be really exciting to me too but I am happy if dual-stack works without having to wait for 1+ minute for DHCPv4 timeouts.

s-hadinger, 2022-11-28:
The only short term need for IPv6 in the short term is Matter support. Matter requires IPv6, and I'm still unclear how it would work on a IPv4 only local network (which is a wide majority of home installations). I'm doubtful that Matter wouldn't work at all on a IPv4 only network.

urbaniak, 2022-11-28:
all matter devices will just use auto configured link local ipv6 addresses even on ipv4 only network and be able to talk to each other that way

Leseratte10, 2022-11-28:
> all matter devices will just use auto configured link local ipv6 addresses even on ipv4 only network and be able to talk to each other that way

This. To communicate with other clients in the same local network, neither your internet connection nor your router needs to support IPv6. All devices automatically have their own address in fe80::/16, based on their MAC address, and just start using it.

MaZe3D, 2022-11-28:
Maybe this is true for Matter Devices but no IPv6 is in 2022 a no go. I have several IPv6 only networks at home and because of Tasmota I need to maintain a separate IPv4 enabled and isolated network. If we can get Tasmota to fe80:: addresses we can get it to support SLAAC to. Then it would only need its prefixes from the routers within the network in order to work.

kroon040, 2022-11-29:
Should be great that ipv6 comes to tasmota. I can also check if SLAAC works, so if there are candidate version, I can test

altelch, 2022-11-29:
What happened to v6, didn't follow for a while but in October 2019 I made some PRs (#6634) to have at least dual stack. There was still some stuff like IP filters missing but the basic functions where there.

s-hadinger, 2022-11-29:
It's mostly there indeed, but Arduino framework has no support for IPv6 on ESP32 contrary to ESP8266. I'm doing brain transplant to bring it to ESP32 as well.

altelch, 2022-11-29:
Uarg...no support in the Framework is bad. Thank you for info and working on it :-)

s-hadinger, 2022-11-29:
Just to keep you updated with progress. IPv6 link-local and DHCPv6 works. I couldn't try SLAAC but it should also work. The device has IPv6 addresses, but I'm not sure it's using it for MQTT and web server. I might need your help for testing for this.

I will push the changes to the repo tomorrow

s-hadinger, 2022-11-30:
IPv6 preview is now merged. For ESP32 you need to self compile with
`platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/1051/framework-arduinoespressif32-release_v4.4-e7f0f8b3f4.zip`

DHCPv6 works fine, SLAAC is untested (unifi doesn't support) it.
Ethernet is not tested.
Web server is not listening on IPv6 global address yet
MQTT on IPv6 address is not tested

https://github.com/arendst/Tasmota/pull/17230

barbudor, 2022-11-30:
I'm totally nood regarding IPv6, but I understand that only dynamic addresse (DHCP or SLAAC)
Is that static addresses doesn't exist in IPv6 or just a limitation of this initial support ?

sfromis, 2022-11-30:
While I can claim zero expertise around IPv6, my strong impression (Google-fu) is that static IP addresses are also possible with IPv6.

techman83, 2022-12-01:
@barbudor by default Privacy and mechanisms to avoid IP clashes without a DHCP server (in the case of SLAAC) make the address random. But on linux you can disable privacy, and set a seed token that creates an address that is consistent beyond the prefix. 

https://www.internetsociety.org/resources/deploy360/2014/privacy-extensions-for-ipv6-slaac/

https://datatracker.ietf.org/doc/id/draft-chown-6man-tokenised-ipv6-identifiers-02.txt

s-hadinger, 2022-12-01:
@barbudor With this PR, Tasmota always has a Link-Local address which is non-routable, and is static because derived solely from the MAC address. It's sufficient to have LAN communication. It uses DHCPv6 and possibly SLAAC to gain a routable address which is also static and derived from MAC address as well. I haven't seen a temporary address discovered by Tasmota, maybe because esp-ids only supports stateless mode (whatever it means)

kroon040, 2022-12-01:
> IPv6 preview is now merged. For ESP32 you need to self compile with `platform_packages = framework-arduinoespressif32 @ [...]

For the esp8266, is it default enabled in the development binaries?

Is it possible that there are already compiled binaries available here for test (esp8266 and esp32)?

https://github.com/Jason2866/Tasmota-specials

barbudor, 2022-12-01:
@s-hadinger my question was more : as of today there is no IPv6Address<x> commands
Is that a gap that would/will need to be filled?
Or is static IP set by user something that IPv6 want to make obsolete?

As of today most of my infra is using static IP (v4) set by myself so wondering if IPv6 would force my to change of religion 😂

Leseratte10, 2022-12-01:
I don't think static IPv6 itself is needed - that would completely break on each prefix change. But it'd be cool to have something like "ip token" on Linux - a way to configure the lower 64-bit of the IPv6 address (usually the EUI64 created from the device MAC). That way you could set the token to like "::1:2:3:4", and that would automatically get appended to the prefix (2001:db8:aaaa:bbbb:1:2:3:4). 

That way you could use a static suffix and still have it work if your router changes the prefix.

kroon040, 2022-12-01:
Hi,

I have compiled it on an esp8266, and run some tests.

Laptop and tamsota device in the **same** vlan:

- SLAAC works it get an ipv6 address. (global address)
- ping works
-  webui listen to the address, use the http://[2001:xxxx:xxxx:xxxx]
- got and ipv6 local (FE80).
-  can't ping local address
- als webui can't ping.

I have default tasmota on **other** vlan than the laptop

- SLAAC works, get an ipv6 in that vlan.
- ping doesn't work, looks like the implementation doesn't use the default gateway from this address?
- webui not working ofcourse (ping also not working no routing)
- got and ipv6 local (FE80).
-  can't ping local address
- als webui can't ping.

s-hadinger, 2022-12-02:
Thanks @kroon040 for testing. It looks in a better shape on ESP8266 than ESP32.

@barbudor Setting a static IPv6 token might be doable with esp-idf, but the Arduino framework is playing against us here, unless we almost redevelop part of it to  introduce IPv6. I'm trying to enable v6 with minimal disruption.

Currently the last 64 bits are stable and derived solely from MAC address, you can consider it static.

kroon040, 2022-12-02:
@s-hadinger , I got it compiled also for esp32, but it looks link that I get form SLAAC an global address, but not yet pingable or whatever. I go to test first both again with full erase and report later this weekend the results for esp8266 and esp32

Jason2866, 2022-12-02:
@kroon040 
> Is it possible that there are already compiled binaries available here for test (esp8266 and esp32)?

Will do if the IPv6 implementation is going to work somehow (experimental).
All provided firmwares have to work before beeing provided precompiled.

s-hadinger, 2022-12-02:
> I got it compiled also for esp32, but it looks link that I get form SLAAC an global address, but not yet pingable or whatever.

Is the link-local addressable pingable?

kroon040, 2022-12-02:
I flashed new devices. It looks like the esp8266 works on the same subnet as also on different subnet. I dont know for sure if the device gets an global adres and the right router prefix if the local link address is still pingable. It looks like de esp32 with SLAAC doesn't work or get the router prefix and that only it works in the same subnet.

New tests **ESP8266**

On  **same** subnet (mqtt / tasmota / laptop)

- [x]  - SLAAC works it get an ipv6 address. (global address)
- [x] - ping works to global address
- [x] - webui listen to the address, use http://[2001:xxxx:xxxx:xxxx]
- [x] - MQTT works with ipv6 to broker on ipv6 (see connected client in log for mosquitto)
- [x] - got and ipv6 local (FE80).
- [ ] - can ping local address (expected behavior?))

On  **other** subnet (mqtt / tasmota / laptop)

- [x]  - SLAAC works it get an ipv6 address. (global address)
- [x] - ping works to global address
- [x] - webui listen to the address, use http://[2001:xxxx:xxxx:xxxx]
- [x] - MQTT works with ipv6 to broker on ipv6 (see connected client in log for mosquitto)
- [x] - got and ipv6 local (FE80).
- [ ] - can ping local address (expected behavior)

Tests ESP32

On **same** subnet (mqtt / tasmota / laptop)

- [x]  - SLAAC works it get an ipv6 address. (global address)
- [x] - ping works to global address
- [x] - webui listen to the address, use http://[2001:xxxx:xxxx:xxxx]
- [x] - MQTT works with ipv6 to broker on ipv6 (see connected client in log for mosquitto)
- [x] - got and ipv6 local (FE80).
- [x] - can ping local address (expected behavior?))

On **other** subnet (mqtt / tasmota / laptop)

- [x]  - SLAAC works it get an ipv6 address. (global address)
- [ ] - ping works to global address
- [ ] - webui listen to the address, use http://[2001:xxxx:xxxx:xxxx]
- [ ] - MQTT works with ipv6 to broker on ipv6 (see connected client in log for mosquitto)
- [x] - got and ipv6 local (FE80).
- [ ] - ping local address (expected behavior))

s-hadinger, 2022-12-02:
@kroon040 I'm surprised that so many features are already working on local network.
How did you test MQTT on same network? Did you just set the IPv6 address via `MqttHost`?

s-hadinger, 2022-12-02:
> Also, are you able to test if the device correctly sets up multiple IPv6 addresses if for example the router announces both global addresses and ULA addresses, and does the web UI / MQTT work over both?

I have setup an IPv6 ssid for testing with DHCP disabled. But it's not for now.

I'm not sure what lwip does with ULA addresses, and I don't see any option on my DHCPv6 to implement it

s-hadinger, 2022-12-02:
@kroon040 Testing http://[2001:xxxx:xxxx:xxxx] doesn't work for me on ESP32. I did some testing with UDP, and it does not listen on IPv6 by default. Can you confirm that it worked for you?

kroon040, 2022-12-02:
> @kroon040 I'm surprised that so many features are already working on local network. How did you test MQTT on same network? Did you just set the IPv6 address via `MqttHost`?

I checked again, and on the ESP8266 it works, and with the ip address in mqtthost (in the webgui). In the ESP32 the web and the mqtt on same subnet fails. I thought it worked but not on the ESP32

kroon040, 2022-12-02:
> @kroon040 Testing http://[2001:xxxx:xxxx:xxxx] doesn't work for me on ESP32. I did some testing with UDP, and it does not listen on IPv6 by default. Can you confirm that it worked for you?

No the web on the esp32 doesn't work, only ping is working. I changed the test results for the esp32

kroon040, 2022-12-02:
Looking at the roadmap  from espressif there are many ipv6 esp32 related issues planned for 3.0.0

https://github.com/espressif/arduino-esp32/milestone/4

I think maybe only esp8266 is ready for ipv6 and not yet esp32?

[47 more replies]

closed 2022-12-17 by s-hadinger