arendst/Tasmota #2522: Can't use GPIO9 and GPIO10 on "Generic" module for ESP8285

Leseratte10, 2018-04-22:
On the ESP8285, the GPIO ports 9 and 10 are not used for anything flash-related and can be used for other things, as seen on the existing ESP8285 devices (4CH, Touch, 4CH Pro). When using the "Generic" module setting, GPIO 9 and 10 can't be selected (as this is intended for an ESP8266 where these can't be used). 

Would it be possible to either make GPIO9 and 10 selectable in the Generic module (however, people may then mess up their ESP8266 when selecting them) or add a second Generic module for the ESP8285 where these are selectable?

(German forum post about this: https://smarthome.schimmer-media.de/forum/index.php?thread/1019-tasmota-mod/ )

nhorvath, 2018-04-24:
If you just want it to work for yourself you can edit sonoff_template.h and change the 0 in that pin location to GPIO_USER for your device. If you think it should be that way for everyone submit a pull request with that change, arendst is pretty good about merging simple things like that.

closed 2018-06-23 by stale[bot]

andrethomas, 2018-10-14:
The reason is that the ESP8266 uses pin 9 for the SPI flash - Perhaps make a new feature request and perhaps the developers will add an option for Generic ESP8285.

vontrapp, 2018-10-14:
I just checked the firmware bin file that I am using. That bin file has byte '6e' for the 2 pins already. Yet still the gui doesn't show them available. Thus I'm not hopeful that even a recompile will make those available.

Is it checking somehow whether it's running on an 8266? If so the check may be wrong. Or there's some other reason those pins aren't being made available...

arendst, 2018-10-14:
There is an undocumented command setoption51.

No support given if you do not know what the consequences are.

vontrapp, 2018-10-14:
I tried setting that to "on" and no effect. Is there a particular value I should set it to?

andrethomas, 2018-10-14:
Setoption51 on and then go to Generic Module

[image]

vontrapp, 2018-10-14:
I'm still on version 6.2.1. Do I need to run the development branch for this?

arendst, 2018-10-14:
Yes.

vontrapp, 2018-10-14:
So... the reason I'm looking for these pins is because I have a Sonoff T1 2CH, which uses GPIO09. Does this mean someone is unprotected from said consequences if they choose "Sonoff T1 2CH" (or 3ch) for the module conig?

andrethomas, 2018-10-14:
Unlikely since it is used as a button on the T1 2CH - other users will not have anything connected to that pin.

vontrapp, 2018-10-14:
except flash...?