arendst/Tasmota #2525: Disabling Pullup resistor for GPIO

nhorvath, 2018-04-22:
I have a PIR sensor that doesn't like the internal pullup on GPIO02 on my Sonoff T1 UK 3CH version. When the PIR goes "off" it is showing about 1.2V and 3.2V when on. The 1.2V is enough voltage for the pin to occasionally consider the input high. I tried adding a pulldown resistor which helped reduce the frequency but it would still happen occasionally.

My solution was to make a code change to switch GPIO02 to INPUT (floating) rather than INPUT_PULLUP. What do you think about adding a user configurable way to do this?
```
  for (byte i = 0; i < MAX_SWITCHES; i++) {
    if (pin[GPIO_SWT1 +i] < 99) {
      pinMode(pin[GPIO_SWT1 +i], (16 == pin[GPIO_SWT1 +i]) ? INPUT_PULLDOWN_16 : ((2 == pin[GPIO_SWT1 +i]) ? INPUT : INPUT_PULLUP));
      lastwallswitch[i] = digitalRead(pin[GPIO_SWT1 +i]);  // set global now so doesn't change the saved power state on first switch check
    }
  }
```

ascillato, 2018-06-06:
Hi,

For all your PIR sensors happened this, or for any in particular?

nhorvath, 2018-06-06:
It happens with all my AM312 based sensors.

abmantis, 2018-08-12:
I'm adding a new device that also needs the GPIO_SWT1 to have no pullup.

ascillato2, 2018-08-23:
Implemented. Closing issue. Please, ask to reopen if needed. Thanks.

closed 2018-08-23 by ascillato2

nhorvath, 2018-08-23:
@ascillato2 can you point me to the documentation on how to use the new config?

ascillato, 2018-08-23:
Yes,

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/_changelog.ino

Now, with last develompent version, you have in the webmenu in the module configuration new options from the dropdown selection.