arendst/Tasmota #2527: How to set the Sonoff module model type without using Webserver?

aaamitsingh, 2018-04-23:
I have different sonoff module like Sonoff 4-ch, pro, sv, pow and I don't want to use the web server so I want to know:

-How to set the Sonoff module model type from code?
-How to set the Sonoff module model type from MQTT?

aaamitsingh, 2018-04-24:
@Frogmore42: Thanks mate any idea how to hardcode this in our code?

Frogmore42, 2018-04-24:
I don't see a way to set the default module type.  It looks like the code is depending on it to default to zero and doesn't explicitly set it.  Settings.ino has the code that sets the default, so you could change it there.

But, I would consider that a generally not best practice.  You would need to keep track of lots of details over an extended period of time to get it right, if you have multiple devices.  If you don't have multiple, I don't believe it would save you any time.

So, why do you want to do this?

Some of the benefit of Tasmota is that you can use the same code on multiple devices.  You can flash new FW and it still remembers it's settings.  This is a huge benefit.  It means you have a one time configuration that you need to do when you deploy a new device.  After that you can deploy the same FW image to all previously configured devices.  Why would you want to have keep track of which FW image goes on which device?

aaamitsingh, 2018-04-24:
You have a valid point and its make sense. Thank a ton

closed 2018-04-24 by aaamitsingh