arendst/Tasmota #2529: Analog0 doesn not come with MQTT

baukeplugge, 2018-04-23:
I'm using the nodeMCU hardware to get a mqtt trigger when there is some change in the reading.
I have commented the USE_ADC_VCC setting
so I can read the anolog0 value.
but this value is not send to the MQTT subscriber.

its also not shown in the console output.

any ideas how to capture this output ?

arendst, 2018-04-23:
How about:

Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``

Works just fine:
```
13:50:54 MQT: tele/ring2/SENSOR = {"Time":"2018-04-23T13:50:54","Analog0":4,"INA219":{"Voltage":4.410,"Current":0.020,"Power":0.087}}
```

baukeplugge, 2018-04-23:
Hi 

the hardware is the NodeMCU
[Example image](https://alselectro.files.wordpress.com/2017/11/image-1.jpg)

not sure how I should get the satus 0 output. 
but this is what is in the start of the console output:
```
12:12:46 MQT: sonoff/AN01-SW-IOT06/SENSOR = {"Time":"2018-04-23T12:12:46","Analog0":0}
12:17:29 APP: Serieel logging uitgeschakeld
12:17:47 MQT: sonoff/AN01-SW-IOT06/STATE = {"Time":"2018-04-23T12:17:47","Uptime":"0T00:10:20","Wifi":{"AP":1,"SSId":"xxxxxx IOT","RSSI":100,"APMac":"xxxxxxxxxx"}}
```

once in the 5 minutes I get this messages in the console 
13:27:55 MQT: sonoff/AN01-SW-IOT06/STATE = {"Time":"2018-04-23T13:27:55","Uptime":"0T01:20:28","Wifi":{"AP":1,"SSId":"xxxx IOT","RSSI":88,"APMac":"xxxxxxx"}}
13:27:55 MQT: sonoff/AN01-SW-IOT06/SENSOR = {"Time":"2018-04-23T13:27:55","Analog0":520}

but im not getting the update on the sonoff/AN01-SW-IOT06/Analog0 for example as soon as there has been a change.

arendst, 2018-04-23:
I see you get the teleperiod update message at 13:27:55 as designed.

You won't get an update when something changes as that would impact the performance of the device massively. You might want to play with command teleperiod.

See wiki regarding command usage and syntax.

I see no problem here.

baukeplugge, 2018-04-23:
thx,

I have checked the TelePeriod now. but the lowest to go is 10 sec. i want to readout the voltage change from my doorbell. so if it is only reading it once per 10 second there would be a big change the doorbell will ring in between.

is it possible to overwrite the 10 sec minimal setting? can I maybe change that in the source code somewhere ?

arendst, 2018-04-29:
With version 5.12.0n you have to possibility to use a rule triggering on any analog change. See Wiki rules example 6.

ascillato, 2018-06-02:
Hi @baukeplugge 

If your issue was solved, please, close it. Thanks :+1:

closed 2018-06-11 by ascillato2