arendst/Tasmota #2531: MQTT enable does not work

vansemtt, 2018-04-23:
Hi,
I am using SonOff Basic with Firmware 5.12.0j. 
Basic config is fine, but trying to enable MQTT in the web interface results in Device not working, providing again WiFi Install network, running slow, much losses of ping, not providing web interface.

Reset of Device is possible with the 40 sec. Button Press.

Perhaps any idea, what is wrong?
Greetings
Matthias

Frogmore42, 2018-04-23:
That config works great for many people.  Please provide the requested information in the issue template.  You probably have a bad config.

ascillato, 2018-04-23:
Hi, seems to be a MQTT config problem.

When using on the user_config.h file:

`#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT`

Makes Tasmota use the pubsubclient library and that library does that hung and slowness if the sonoff can't connect to a MQTT Server.

Please, recompile using the Tamosta MQTT Library, commenting the line 188 and uncommenting the line 190:

`#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT`

Please, tell us if that works for you.

vansemtt, 2018-04-24:
Hi,
thank you very much for your attention. Herby the output of the command status 0:
06:34:55 CMD: Status 0
06:34:55 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
06:34:55 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"","RestartReason":"Software/System restart","Uptime":"0T01:42:19","StartupUTC":"2018-04-24T03:52:36","Sleep":0,"BootCount":5,"SaveCount":21,"SaveAddress":"F7000"}}
06:34:55 RSL: STATUS2 = {"StatusFWR":{"Version":"5.12.0j","BuildDateTime":"2018.04.12 14:51:08","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
06:34:55 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ",""],"TelePeriod":300,"SetOption":["00008001","55818000"]}}
06:34:55 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
06:34:55 RSL: STATUS5 = {"StatusNET":{"Hostname":"SonOff-04d","IPAddress":"192.168.188.60","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.188.1","Mac":"DC:4F:22:2D:31:72","Webserver":2,"WifiConfig":2}}
06:34:55 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 24 05:34:55 2018","Local":"Tue Apr 24 06:34:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:42","Sunset":"19:54"}}
06:34:55 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-04-24T06:34:55"}}
06:34:55 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-04-24T06:34:55","Uptime":"0T01:42:19","Vcc":3.427,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FRITZ","RSSI":52,"APMac":"C8:0E:14:D0:C9:10"}}}

This is before switching MQTT on.

Jason2866, 2018-04-24:
06:34:55 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-04-24T06:34:55"}}
06:34:55 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-04-

RSL messages appears only if there is no mqtt connection.
What messages do you get in your logs if mqtt is up?

vansemtt, 2018-04-25:
Hi @Jason2866, That's actual my problem. If I enable MQTT, I do not have access to the son off anymore. Even the IP Adress is no longer reachable in my network.

Jason2866, 2018-04-25:
Try instead of SDK 2.4.0 -> 2.3.0. Many users have problems with 2.4.0!