arendst/Tasmota #2533: Can't stop Sonoff SV relay triggering after power restart

ceraz68, 2018-04-23:
I use a Sonoff SV to emulate a push button for a garage door successfully integrated to homebridge. My problem is that the Sonoff relays triggers after power restart, I unplug the usb power cable and re-plug to simulate a power failure. Once replugged the relay immediately triggers and opens the door, I don't want to happen ie don't trigger after power-up.

My MQTT broker is running as synology package

Sonoff GPIOs not used
Onboard resistors removed to isolate power to the esp
Garage opener wires connected to positive contacts of the input and output
SwitchMode1 set to 0
SwitchMode2 set to 2
Pulsetime set to 10

I tweaked PowerOnState from 3 to 0 => no impact
I tweaked SaveState (now named SetOption0) from 1 to 0 => no impact

Any ideas?
Thanks

Console output (with dummy SSID/IP)
===========================
00:00:00 Project sonoff SonoffSV (Topic sonoffsv, Fallback DVES_9596DC, GroupTopic sonoffs) Version 5.9.1
00:00:00 WIF: Connecting to AP1 MyWifiLAN in mode 11N as sonoffsv-5852...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoffsv-5852.local with IP address 192.168.X.X
20:09:25 MQT: Attempting connection...
20:09:25 MQT: Connected
20:09:25 MQT: tele/sonoffsv/LWT = Online (retained)
20:09:25 MQT: cmnd/sonoffsv/POWER = 
20:09:25 MQT: tele/sonoffsv/INFO1 = {"Module":"Sonoff SV", "Version":"5.9.1", "FallbackTopic":"DVES_9596DC", "GroupTopic":"sonoffs"}
20:09:25 MQT: tele/sonoffsv/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoffsv-5852", "IPAddress":"192.168.X.X"}
20:09:25 MQT: tele/sonoffsv/INFO3 = {"RestartReason":"Power on"}
20:09:27 MQT: stat/sonoffsv/RESULT = {"POWER":"OFF"}
20:09:27 MQT: stat/sonoffsv/POWER = OFF
20:09:34 MQT: tele/sonoffsv/STATE = {"Time":"2018-04-23T20:09:34", "Uptime":0, "Vcc":3.136, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"MyWifiLAN", "RSSI":100, "APMac":"11:11:11:11:11:11"}}

Console Output for Status 0
=====================
20:37:28 CMD: status 0
20:37:28 MQT: stat/sonoffsv/STATUS = {"Status":{"Module":3, "FriendlyName":"SonoffSV", "Topic":"sonoffsv", "ButtonTopic":"0", "Power":0, "PowerOnState":0, "LedState":1, "SaveData":1, "SaveState":0, "ButtonRetain":0, "PowerRetain":0}}
20:37:28 MQT: stat/sonoffsv/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":17, "SaveCount":46, "SaveAddress":"F6000"}}
20:37:28 MQT: stat/sonoffsv/STATUS2 = {"StatusFWR":{"Version":"5.9.1", "BuildDateTime":"2017-11-13T21:43:41", "Boot":6, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
20:37:28 MQT: stat/sonoffsv/STATUS3 = {"StatusLOG":{"SerialLog":2, "WebLog":2, "SysLog":0, "LogHost":"domus1", "LogPort":514, "SSId1":"MyWifiLAN", "SSId2":"indebuurt2", "TelePeriod":300, "SetOption":"55800008"}}
20:37:28 MQT: stat/sonoffsv/STATUS4 = {"StatusMEM":{"ProgramSize":471, "Free":532, "Heap":25, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
20:37:28 MQT: stat/sonoffsv/STATUS5 = {"StatusNET":{"Hostname":"sonoffsv-5852", "IPAddress":"192.168.X.X", "Gateway":"192.168.X.X", "Subnetmask":"255.255.255.0", "DNSServer":"W.X.Y.Z", "Mac":"11:11:11:11:11:11", "Webserver":2, "WifiConfig":2}}
20:37:28 MQT: stat/sonoffsv/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.X.X", "MqttPort":1883, "MqttClientMask":"DVES_%06X", "MqttClient":"DVES_9596DC", "MqttUser":"user", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
20:37:28 MQT: stat/sonoffsv/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 23 19:37:28 2018", "Local":"Mon Apr 23 20:37:28 2018", "StartDST":"Sun Mar 25 02:00:00 2018", "EndDST":"Sun Oct 28 03:00:00 2018", "Timezone":1}}
20:37:28 MQT: stat/sonoffsv/STATUS10 = {"StatusSNS":{"Time":"2018-04-23T20:37:28"}}
20:37:28 MQT: stat/sonoffsv/STATUS11 = {"StatusSTS":{"Time":"2018-04-23T20:37:28", "Uptime":0, "Vcc":3.149, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"MyWifLAN", "RSSI":100, "APMac":"11:11:11:11:11:11"}}}

Frogmore42, 2018-04-24:
You are running a really old version, but that is not likely the issue.  Usually this type of problem is caused by retain messages on your mqtt broker or the home automation system.  What home automation are you using?

ceraz68, 2018-04-24:
Thanks, I didn't pay attention to the version, I used SONOATA to flash the Sonoff SC and assumed the latest would be installed.

I'm running homebridge-mqtt-switch-tasmota on a homebridge docker on a synology NAS. The mqtt broker is the mosquito package for sinology (on the same synology).

I read some posts on mqtt retain - didn't really understand what it means. I doubt the broker is sending an mqtt power on request as the garage door opens instantly once the SV is powered. I would imagine the time the SV connects to wifi and "binds" with the mqtt broker would take 1-2 seconds.

Jason2866, 2018-04-24:
20:09:25 MQT: tele/sonoffsv/INFO3 = {"RestartReason":"Power on"}
20:09:27 MQT: stat/sonoffsv/RESULT = {"POWER":"OFF"}
20:09:27 MQT: stat/sonoffsv/POWER = OFF

This indicates that your broker sends the commands. So your search has to be in that direction.
I assume that messages a stored in your broker  (retain). This messages are resend after the device is reachable again. Try to get help / infos from the homebridge community. I cant help here, i dont use that software.

Frogmore42, 2018-04-27:
Change the MQTT name of the device and see what happens.  I don't use homebridge or homeassistant.  Given how frequently people who do seem to have problems I am not inclined to try.  It seems pretty clear that you have something configured incorrectly (which might be a default from something you installed) so starting over might make sense.

MMMM78, 2018-04-29:
I had more or less the same problem while restarting HA (at every restart of HA, it triggered my garagedoor-i use the same module).  I solved it by setting ‘retain’ to ‘false’ in my configuration.yaml file. Maybe it’s of help for you too.

ceraz68, 2018-06-17:
I went quiet for a while as I bricked the sonoff SV when trying to upgrade the firmware to the latest tasmota (continuous red LED). I ordered a new one and performed the below steps.

After power cycling (unplugging/plugging power to simulate a power failure), the relay no longer triggers, good!

I no longer get the immediate message "17:47:45 MQT: homeassistant/switch/sonoffsv/config = (retained)"

Nothing has changed on my MQTT and homebridge server setup. Compared to the previous sonoff I did not configure SwitchMode1 nor SwitchMode2. In view of the failure to upgrade the sonoff SV firmware, I put my bug down to non-pefect Sonoff SV !!

Thanks for your support in this forum.

New sonoff with fw 1.5.5
Used Sonota installs tasmota: it installs tasmota 5.9.1
Configure mqtt on sonoff and choose a new topic to avoid interference with previous
Upgrade to tasmota version 5.14
* checked power-off-power-on: relay does not trigger
Tested operation of sonoff SV through iOS HomeKit 
* checked power-off-power-on: relay does not trigger
No trace of homeassistant message in console 
Changed Pulsetime to 10
Tested operation of sonoff SV through iOS HomeKit 
* checked power-off-power-on: relay does not trigger
No trace of homeassistant message in console

Frogmore42, 2018-06-17:
It sounds like everything is working.  If so please close this issue.

closed 2018-06-18 by ceraz68