arendst/Tasmota #2536: Many warnings during compoilation for  espressif8266@1.7.0 ; v2.4.1 PIO

roblad, 2018-04-24:
Hi,

I discovered that new release of PIO latest release Atom 1.26 x64 Platform-IDE 2.1.4

platform = espressif8266@1.7.0 ; v2.4.1

during compilation reports several warnings i.e.
In function Void MqttPublishDomotichPowerState (byte)';
"format '%d' expects argument .... but argument 4 has long unsigned int 

many of them are in xdrv_09_timers.ino 

settings

rooles etc.

Jason2866, 2018-04-24:
I am interested. What is the benefit of the new version? Bug fixes? New features?

reloxx13, 2018-04-24:
1.7.0 is still in progress

see here: https://github.com/platformio/platform-espressif8266/issues/76

@Jason2866  changelog: https://github.com/esp8266/Arduino/releases/tag/2.4.1

closed 2018-04-26 by arendst