arendst/Tasmota #2537: BE_MINIMAL breaks tls

or78rXxZ, 2018-04-24:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
sonoff basic

- [ ] Provide the output of command ``status 0``

Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff_b6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoff_b6","OtaUrl":"http://grumpy3.grumpy-net:9999/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:02:07","StartupUTC":"2018-04-24T13:47:03","Sleep":0,"BootCount":12,"SaveCount":29,"SaveAddress":"FB000"}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS2 = {"StatusFWR":{"Version":"5.12.0m-minimal","BuildDateTime":"2018-04-24T08:45:10","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":4,"LogHost":"grumpy3.grumpy-net","LogPort":514,"SSId":["dzc_ntA61Fvgbhit_2","dzc_ntA61Fvgbhit"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS4 = {"StatusMEM":{"ProgramSize":413,"Free":588,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS5 = {"StatusNET":{"Hostname":"sonoff-b6","IPAddress":"192.168.2.30","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"EC:FA:BC:87:3D:0C","Webserver":2,"WifiConfig":4}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS6 = {"StatusMQT":{"MqttHost":"grumpy2.grumpy-net","MqttPort":8883,"MqttClientMask":"sonoff_b6","MqttClient":"sonoff_b6","MqttUser":"sonoff_b6","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 24 13:49:10 2018","Local":"Tue Apr 24 08:49:10 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-5}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS10 = {"StatusSNS":{"Time":"2018-04-24T08:49:10"}}
Apr 24 08:49:10 sonoff-b6 ESP-RSL: stat/sonoff_b6/STATUS11 = {"StatusSTS":{"Time":"2018-04-24T08:49:10","Uptime":"0T00:02:07","Vcc":3.560,"POWER":"OFF","Wifi":{"AP":1,"SSId":"dzc_ntA61Fvgbhit_2","RSSI":58,"APMac":"0E:41:58:00:9A:B5"}}}

- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

I compiled 5.12.0m with tls and no web server. Everything appears to work.
I only added #define BE_MINIMAL to user_config_override.h and recompiled.
The device is unable to connect to mqtthost.

08:48:29 MQT: Attempting connection...
[hostByName] request IP for: grumpy2.grumpy-net
[hostByName] Host: grumpy2.grumpy-net IP: 192.168.2.16
:ref 1
:wr 73 73 0
:wrc 73 73 0
:sent 73
:er -9 0x00000000
08:48:44 MQT: Connect failed to grumpy2.grumpy-net:8883, rc -4. Retry in 10 sec

Mosquitto log shows

1524577683: New connection from 192.168.2.30 on port 8883.
1524577683: OpenSSL Error: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
1524577683: Socket error on client <unknown>, disconnecting.

Leseratte10, 2018-04-24:
Well that's the point of BE_MINIMAL - remove all features, including TLS, so the image is as small as possible. You are supposed to immediately update to a non-minimal version after flashing.

or78rXxZ, 2018-04-24:
With no web server I initiate upgrades with mqtt but with no tls mqtt does not work so I can not upgrade after I install the minimal image.

nhorvath, 2018-04-24:
Yes, an ssh tunnel would be a good option. Unfortunately, it's too much to ask of this inexpensive little chip to run a full TLS stack and do much of anything else.

closed 2018-04-29 by arendst