arendst/Tasmota #2538: Sonoff not reachable with IP but MQTT is runing

mated77, 2018-04-24:
Hi. I now have a couple of Sonoffs but this time i don´t know what to do.

It is a Sonoff RF and i flashed it with success. First with 5.11. also trying 5.12. and the minimal Versions of it. Every time it is the same issue: i can not connect to the device. Browser is loading and says "can´t connect after a few minutes". Sometimes i can connect for a short time but when i want to go to the settings or logging it stuck again. I also try a ping test. Most times no connection or after a few pings it is gone -> https://i.imgur.com/dpROwFr.jpg . What really makes me wonder is the fact that all results are coming to the MQTT Broker. So, the device is connected to my network and to MQTT without problems. I try it without MQTT but this doesn't help (thought it is a overload / memory issue). Someone had the same problem or a solution?

Jason2866, 2018-04-24:
Try a complete reset of the device. Sometimes bad configs are left. (4 sec button press)
Which version of IDE do you use?  2.4.0. is not always working, try 2.3.0
uncomment in platformio.ini:
platform = espressif8266@1.5.0  ; v2.3.0
Restart your mqtt broker. Yes this can help! My broker had made issues, after restart the all where gone

doingnz, 2018-04-24:
try connect from a mobile phone. I cannot get Chrome on PC to connect to IP address, but works fine from Chrome on Android phone!

curzon01, 2018-04-24:
@doingnz - having the opposite situation ;-) lib 2.4.0 works on my PC chrome but not with an Huawei Android Chrome - means there is no general way. My common suggestion - try a _different_ browser platform

doingnz, 2018-04-25:
@curzon01 What is odd is the HTTP GET API works fine on PC, but I cannot see web pages. There must be something very simple wrong that is confusing the clients ... tantalizingly close... yet so far from being a reliable system .

doingnz, 2018-04-25:
I have just tried with v2.3.0
```
[env:sonoff]
platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200
```
that works from both PC and phone

Sigma95, 2018-11-27:
Hi,

I am experiencing the same issue when put my Sonoff S20 far from my wifi router.

The sonoff is not reachable by web browser and don’t appear on the router UI. But it is still toggling when I publish Through MQTT.

Probably the problem is that the WiFi signal is not powerful enough for IP communication but still enough for a lightweight protocol such as MQTT?

curzon01, 2018-11-28:
> Probably the problem is that the WiFi signal is not powerful enough for IP communication but still enough for a lightweight protocol such as MQTT?

This makes no sense: MQTT - as lightweight as is - is using TCP IP commuincation. 

I assume another reason for this behaviour. if possible use network tools to identify the problem:
- is ping to S20 working  (from mqtt server location if possible and from browser location)
- what's traceroute result (from mqtt server location if possible and from browser location)
- try nmap to S20
- if MQTT works request STATUS 0 and have a look for the values IP, RSSI...
- which router you are using and why S20 is not visible