arendst/Tasmota #2539: Is it possible to BLINK OFF circuits that are always on based on ping?

poldim, 2018-04-24:
I’m trying to figure out if I can solve my modem reset circuit with a Sonoff.
Specifically, I have a remote location that has an ISP with poor reliability.  I’d like to implement a solution that looks to ping an IP and would blink the relay off (for _t_ seconds) when it could not reach this IP after (_x_ time or _y_ attempts).  The key thing is that the circuit must default to an ON state if something goes awry.  

Hardware would be a Basic or possibly a powerstrip like Tonbux.

poldim, 2018-04-25:
> This is a scenario for a HA (Home Automation) System. You could do it (for
> example) with Open Hab.
> With a HA you can check "things" and "fire" actions. A complete setup runs
> fine on a Rpi3
> Maybe Domoticz does the job too. Just look around for different HA Systems
> and choose one you like...

I'm running HA but do not want to do with another system.  One of the items
that is being monitored would be the Rpi hosting HA, so cant have it
monitor itself, hence want to do it in tasmota.

Jason2866, 2018-04-25:
I understand. Cant you do it via GPIO pin from the RPi?
I dont use HA so i am asking. For OpenHab there is the possibility to get access to the GPIOs
and if net is lost the router could be reseted via OpenHab this way....

Frogmore42, 2018-04-26:
The simple answer is that Tasmota does not currently do what you want and was not designed for that type of task.

The less simple answer is that it is a small matter of programming to make Tasmota do what you want.  Since it is open source, you can make it do just about anything.

The more complex answer is that Tasmota makes design decisions that are good for it's intended uses, but probably not so good for what you want.

It would take some time to understand the design and then more to modify it to support your needs.

Implementing a good watchdog is actually quite hard.  Implementing a mediocre one is easy 😀

closed 2018-06-25 by stale[bot]