arendst/Tasmota #2540: H801 Dimmer sends out two states

soprater, 2018-04-24:
I'm using the H801 combined with Home Assistant to control my led strips.
When using the dimmer the console of the H801 (and mqtt) sends out two states:
``22:02:08 MQT: stat/sonoff_ledwin/RESULT = {"POWER":"ON","Dimmer":16,"Color":"0622000000","CT":499}
22:02:08 MQT: stat/sonoff_ledwin/RESULT = {"POWER":"ON"}``

Afterwards Home Assistants logs errors when it tries to parse the second state:
``Error parsing value: 'dict object' has no attribute 'Dimmer' (value: {"POWER":"OFF"}``
``ValueError: could not convert string to float: '{"POWER":"ON"}'``

Is it possible to turn off the second state or am I missing something?

My HA config:
````
light:
 platform: mqtt
    name: tasmota_win
    state_topic: "stat/sonoff_ledwin/POWER"
    command_topic: "cmnd/sonoff_ledwin/POWER"
    color_temp_state_topic: "stat/sonoff_ledwin/CT"
    color_temp_command_topic: "cmnd/sonoff_ledwin/CT"
    #state_value_template: "{{ value_json.state }}"
    brightness_state_topic: "stat/sonoff_ledwin/RESULT"
    brightness_command_topic: "cmnd/sonoff_ledwin/Dimmer"
    brightness_value_template: "{{ float(value_json.Dimmer) }}"
    brightness_scale: 100
    qos: 1
````

- [x] What hardware you are using
H801
- [x] Provide the output of command ``status 0``
````
22:14:02 MQT: stat/sonoff_ledwin/STATUS = {"Status":{"Module":20,"FriendlyName":"Sonoff","Topic":"sonoff_ledwin","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0 00:36:51","Sleep":0,"BootCount":17,"SaveCount":195,"SaveAddress":"FA000"}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018.04.24 21:46:39","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00008009"}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS4 = {"StatusMEM":{"ProgramSize":458,"Free":544,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS5 = {"StatusNET":{"Hostname":"sonoff_ledwin-4679","IPAddress":"x.x.x.x","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:6F:92:47","Webserver":2,"WifiConfig":3}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.109","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6F9247","MqttUser":"homeassistant","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 24 21:14:02 2018","Local":"Tue Apr 24 22:14:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS10 = {"StatusSNS":{"Time":"2018.04.24 22:14:02"}}
22:14:02 MQT: stat/sonoff_ledwin/STATUS11 = {"StatusSTS":{"Time":"2018.04.24 22:14:02","Uptime":"0 00:36:51","Vcc":3.437,"POWER":"OFF","Dimmer":16,"Color":"0622000000","CT":499,"Scheme":0,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":98,"APMac":"30:B5:C2:81:5B:AC"}}}
````

SecurityAddicted, 2018-04-25:
I had the same issue. Resolved by not using at all the POWER attribute, just use Dimmer. When setting Dimmer to 0 the device turns off automatically, and back on when Dimmer > 0.

soprater, 2018-04-27:
When I only use the brightness slider I'll get far too much mqtt messages. (1x off, 2x on (with dimmer))

````
11:47:16 MQT: stat/sonoff_ledwin/RESULT = {"POWER":"OFF"}
11:47:16 MQT: stat/sonoff_ledwin/POWER = OFF
11:47:19 MQT: stat/sonoff_ledwin/RESULT = {"POWER":"ON","Dimmer":20,"Color":"1D15000000","CT":499}
11:47:19 MQT: stat/sonoff_ledwin/RESULT = {"POWER":"ON"}     <-------
11:47:19 MQT: stat/sonoff_ledwin/POWER = ON
11:50:15 MQT: stat/sonoff_ledwin/RESULT = {"POWER":"ON","Dimmer":57,"Color":"533D000000","CT":499}
11:50:15 MQT: stat/sonoff_ledwin/RESULT = {"POWER":"ON"}     <-------
11:50:15 MQT: stat/sonoff_ledwin/POWER = ON
````
Marked messages are resulting the error.
````
Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...at 0x66286c70>)
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/mqtt/__init__.py", line 647, in _mqtt_handle_message
    msg.topic, payload, msg.qos)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 248, in async_run_job
    target(*args)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/light/mqtt.py", line 247, in brightness_received
    device_value = float(templates[CONF_BRIGHTNESS](payload))
ValueError: could not convert string to float: 
````

BenNeumeister, 2018-06-10:
Enable SetOption4 which then provides the responses on messages other than 'Result'.

Then just update your MQTT to match;
i.e.
```
 platform: mqtt
    name: "Kitchen Island Bench"
    effect_list:
      - 0
      - 1
      - 2
      - 3
      - 4
    state_topic: "stat/islandBench/POWER"
    state_value_template: "{{ value }}"
    command_topic: "cmnd/islandBench/POWER"
    brightness_state_topic: "stat/islandBench/DIMMER"
    brightness_command_topic: "cmnd/islandBench/Dimmer"
    brightness_value_template: "{{ value_json.Dimmer }}"
    brightness_scale: 100
    rgb_state_topic: "stat/islandBench/COLOR"
    rgb_command_topic: "cmnd/islandBench/Color2"
    rgb_value_template: "{{(value_json.Channel[0]*2.55)|int}},{{(value_json.Channel[1]*2.55)|int}},{{(value_json.Channel[2]*2.55)|int}}"
    effect_command_topic: "cmnd/islandBench/Scheme"
    effect_state_topic: "stat/islandBench/SCHEME"
    effect_value_template: "{{ value.Scheme }}"
    retain: true
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"_
```

closed 2018-06-12 by ascillato2