arendst/Tasmota #2541: Request for adding domotich multi DS sensor fields adding to web interface when mor then 1

roblad, 2018-04-25:
Hi,

Could you add more fields for domoticz usage for sensors when the same sensor i.e. ds exists for set idx for each sensor through web interface?

MQT: domoticz/in = {"idx":345,"nvalue":0,"svalue":"29.12","Battery":200,"RSSI":10}

arendst, 2018-04-27:
See the latest Rules addition released today allowing to compose rules for every sensor and sending it to Domoticz as you like.

See https://github.com/arendst/Sonoff-Tasmota/wiki/Rules example 5

closed 2018-06-27 by stale[bot]

Dylantje, 2018-08-30:
Dear all
Please a someone have a simple how to?
I cannot understand where to type the rule?
And when i use multiple dallas how do i now which sensor i use? is there a id or something? 

I see the console only do i need to type everytime the code for reading multiple sensors? I set in the skect the multiple reading to on..

Is there a way to start? First all the sensors and then the command? or is tasmota reading all the sensor there are when i plugin them?
And how do i get the sensors vissable in Domoticz? [ i know the dummy and the ix etc ] I mena how is tasmota upload the dallas to domoticz? I have the switch working so the sonoff have conection with domoticz.

Hope 
Thanks!!