arendst/Tasmota #2542: Southern Hemisphere daylight saving

ajaypala, 2018-04-25:
I don't think the southern hemisphere daylight saving code is working correctly or I have done something incorrectly. The local time (and sun rise/set time) is reported as 1 hour ahead of what it should be. UTC time, StartDST and EndDST is being reported correctly. Summer time is +11h and winter time is +10h. It is currently winter. This issue is common across all my Tasmota devices running various different firmware versions.

```
21:32:17 MQT: house/bed3/lamp2/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 25 10:32:17 2018","Local":"Wed Apr 25 21:32:17 2018","StartDST":"Sun Oct 07 02:00:00 2018","EndDST":"Sun Apr 01 03:00:00 2018","Timezone":99,"Sunrise":"07:25","Sunset":"18:20"}}

21:37:34 MQT: house/bed3/lamp2/stat/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-25T17:56:48","Boot":4,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
```
```
// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST  South, First, Sun, Oct, 2, +660  // Southern Hemisphere, First sunday in October at 02:00 +660 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD  South, First, Sun, Apr, 3, +600   // Southern Hemisphere, First sunday in April 03:00 +600 minutes
```

JueBag, 2018-04-25:
The last lines which are supposed to set the DST or STD  dates are commented out!

JueBag, 2018-04-25:
Do you have Timezone=99 in order to DST?

ajaypala, 2018-04-25:
@JueBag The DST and STD lines are not commented out. Timezone is set to 99 as seen in StatusTIM.

arendst, 2018-04-26:
You might have to play with swapped TIME_DST and TIME_STD offsets and possibly StartDST and EndDST.

ajaypala, 2018-05-05:
After testing trying 7 different variations I have found the correct settings for southern hemisphere. The Start and finish dates need to be swapped, but do not swap the offsets. Hopefully this will provide enough info to correct the code.
```
// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST  South, First, Sun, Apr, 3, +660  // Southern Hemisphere, First sunday in October at 02:00 +660 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD  South, First, Sun, Oct, 2, +600   // Southern Hemisphere, First sunday in April 03:00 +600 minutes
```

ascillato2, 2018-06-12:
Bug Solved in 5.14. Thanks for reporting.

closed 2018-06-12 by ascillato2