arendst/Tasmota #2543: Sonoff Pow energy_kWhtoday overflow

vitek499, 2018-04-25:
Sonoff Pow is is designed to 16A. This means the flow until 3500W. If theoretically we plug in a device with maximum sampling, total daily value may be 84kW (3,5*24).

But in code, total conuter is assigned as:
```
unsigned long energy_kWhtoday
```
It means maximal value 4,294,967,295. But energy_kWhtoday is not saved as kW. It is recalculated 
 to kW by devide 100000000. This is visible on the line 85, file `xdrv_03_energy.ino`. If you calculate maximal daily value, its just 42,9kW - which is half the maximum theoretical measurable value.

For example, if you measure the consumption of the refrigerator, you measure absurd values cause the value overflows.

curzon01, 2018-04-25:
Be aware that in the code line you mentioned the unsigned long value deviced by 100000000 is cast to a float so that's enough to store the `energy_kWhtoday` value.

Note: You are mixing physical values - kWh (energy, electric work) is not kW (electric power).
So the maximum possible value on a Pow for electric work are 84 kWh per day, not  84 kW per day so `energy_kWhtoday` stores (what's the var name say) the today energy in kWh.

vitek499, 2018-04-25:
You are right, I'm mixing kWh with kW but it does not change anything on the problem that the maximum theoretical value is 84kWh.

`energy_kWhtoday` is divided when it is passed to the `energy_kWhtotal` or before it is displayed. 

```
void EnergyUpdateToday()
{
  RtcSettings.energy_kWhtoday = energy_kWhtoday;
  energy_daily = (float)energy_kWhtoday / 100000000;
  energy_total = (float)(RtcSettings.energy_kWhtotal + (energy_kWhtoday / 1000)) / 100000;
}
```

`RtcSettings.energy_kWhtoday` is unsigned long, so after energy_kWhtoday is more than 4,294,967,295 (42,9kWh) overflows. This is a value that is not difficult to achieve.

Frogmore42, 2018-04-25:
I agree that there is the possibility of an overflow of the number, but I also agree that there would be a corresponding overflow of the POW, i.e. it would probably melt if it actually carried that much energy.

You must have a very big/inefficient refrigerator if you are coming close to overflowing the number.

vitek499, 2018-04-26:
I changed unsigned long to uint64_t and it works fine. This is considered a temporary solution, not too economical to the device memory. If we are going to talk about good firmware functionality, there should be no such problem, no matter how likely this limit is.

The device runs almost with a continuous load of 3kW for ten days and dont shows aby problems.

Frogmore42, 2018-04-27:
I'm glad it is working for you.  Some/many of the early ones had an issue where they would overheat if they that much current flowing through them for any length of time.  For measuring power of that magnitude I use current transformers.  I also wouldn't trust the connectors and PCB traces with that much current for very long.

As far as the change you made goes, yeah you need a larger variable to hold that much usage and maintain accuracy.  Using the extra 4 bytes of RAM is not an issue.  The question is how did you handle the storage in the RTC ram memory.  That memory is very constrained.  I am not sure exactly when it gets saved to RTC (looks like when settings change and it needs to reboot, but there might be other cases too).

closed 2018-05-03 by arendst